MarlinFirmware/Marlin #21279: [BUG] wrong file format being uploaded into MCU memory (STM32) by VSCode / PIO

KLOSTERS, 2021-03-07:
### Bug Description
Hardware: Bigtreetech SKR MINI E3 V2.0 (STM32F103RCT6) + TFT24 Display (serial connection)

A couple of days ago I somehow killed my Board via SD-Update.
In order to save it back to live i tried to update it via VSCode / PIO (openocd) using STLink.
When I upload the compiled firmware onto the board using STLink, the board remains dead (no USB Connection, "No Printer attached" on the TFT24).
After some research i readed the MCU flash memory back using ST Visual Programmer and found out, that the first chars were **.ELF**. 
STVP also tells me, that the MCU is not programmed.
STVP also tells me, that the *.bin file generated by VSCode is not an Intel Hex Format, when I try to uplad it "manually" to the board.

- tested with bugfix-2.0.x (Downloaded on 06.03.2021)
- no similar bug-reports on the internet

### Configuration Files

Zip File included:
[Marlin-Bugfix-2.0.x BTT_SKR_Mini_E3 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6097213/Marlin-Bugfix-2.0.x.BTT_SKR_Mini_E3.Configs.zip)

### Steps to Reproduce

1. Build the Firmware in VS Code / PIO
2. Upload Firmware via VS Code / PIO and STLink
3. read MCU Memory back via STVP

**Expected behavior:**

Readback of the MCU memory via STVP shows **no** ".ELF" as first bytes.
SD Card Bootloader works.
Board becomes recognized by a PC as a USB Serial Device, 
"No Printer attached" on TFT24 disappears after a short period of time.

**Actual behavior:**

Readback of the MCU memory via STVP **shows** ".ELF" as first bytes.
SD Card Bootloader does not work.
Board becomes not recognized by a PC as a USB Serial Device.
"No Printer attached" on TFT24 does not disappear.

#### Additional Information
**Flash Content**
[image]

**Error Message STVP**
[image]

KLOSTERS, 2021-03-07:
Summary: VSCode puts whole *.elf File to the MCU, instead of machine code it contains.

ellensp, 2021-03-09:
why did you include makefile? its not used in PIO
Can you attach platfomio.ini file?

ellensp, 2021-03-09:
The default build environment for a BOARD_BTT_SKR_MINI_E3_V2_0 are not setup for stlink uploads.
The environment is designed for most users, ie copy firmware.bin to sdcard and reboot controller.

User using stlink are expected to know a bit about why they are doing.

You need to an upload command to your build environment like the following example.

upload_command    = "$PROJECT_PACKAGES_DIR/tool-stm32duino/stlink/ST-LINK_CLI.exe" -c SWD -P "$BUILD_DIR/firmware.bin" 0x8007000 -Rst -Run

Unfortunately this mean you have blown away the bootloader for this board that was at 0x8000000-0x8006FFF

ellensp, 2021-03-09:
Luckily someone has made a backup and put it online https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader/tree/main/bootloader_bin/backed_up_original_bootloaders/SKR%20mini%20E3%20V2.0

thisiskeithb, 2021-03-09:
> The default build environment for a BOARD_BTT_SKR_MINI_E3_V2_0 are not setup for stlink uploads.

The bootloader offset could probably be added similarly to what was done for the SKR Pro and BTT002: https://github.com/MarlinFirmware/Marlin/pull/17657

KLOSTERS, 2021-03-09:
Thanks guys, i appreciate your help! Will definetely try it this weekend. 
Today i ran into a different issue with the ST-Link. I hope my board has no counterfeit chip. 

Here is the platformio.ini
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/6111628/platformio.zip)

KLOSTERS, 2021-03-10:
thanks for your help,
flashing the bootloader (*.bin) directly via ST-Link and uploading the compiled firmware via SD did the trick. The Board is resposive now! 

Funny, but when i modify platformio.ini for flashing the *.bin VSCode it only does generate an *.elf and complains, that there is no *.bin available. 

here is the modified platformio.ini
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/6118110/platformio.zip)

If you like, you may close the issue.

closed 2021-03-11 by ellensp