MarlinFirmware/Marlin #21280: [BUG] Marlin compile failed at PID DEBUG

gummy-tech, 2021-03-07:
### Bug Description

Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\gcode\temp\M303.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\gcode\units\M82_M83.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\lcd\dogm\lcdprint_u8g.cpp.o
Compiling .pio\build\BIGTREE_GTR_V1_0\src\src\lcd\dogm\marlinui_DOGM.cpp.o
Marlin\src\gcode\temp\M303.cpp: In static member function 'static void GcodeSuite::M303()':
Marlin\src\gcode\temp\M303.cpp:54:27: error: 'pid_debug_flag' was not declared in this scope; did you mean 'marlin_debug_flags'?
   54 |       serialprintln_onoff(pid_debug_flag);
      |                           ^~~~~~~~~~~~~~
      |                           marlin_debug_flags
*** [.pio\build\BIGTREE_GTR_V1_0\src\src\gcode\temp\M303.cpp.o] Error 1
==================================================================================== [FAILED] Took 150.06 seconds ====================================================================================

Environment       Status    Duration
----------------  --------  ------------
BIGTREE_GTR_V1_0  FAILED    00:02:30.060
================================================================================ 1 failed, 0 succeeded in 00:02:30.060 ================================================================================
The terminal process "C:\Users\Acer\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\Acer\.platformio\penv\Scripts\platformio.exe run <

Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6097367/Marlin.zip)

ellensp, 2021-03-07:
Using provided configs does not have PID_DEBUG enabled and does not produce the error. Not very useful.

However when PID_DEBUG is enabled the error occurs.

Marlin/src/gcode/temp/M303.cpp
change       serialprintln_onoff(.pid_debug_flag); 
to       serialprintln_onoff(thermalManager.pid_debug_flag);

ellensp, 2021-03-07:
PR to fix this created

ellensp, 2021-03-08:
This has been merged into bugfix. Closing.

closed 2021-03-08 by ellensp