MarlinFirmware/Marlin #21281: [BUG] White color used after PRINTER_EVENT_LEDS is incorrectly using W which shouldn't be valid with NeoPixels

lightmaster, 2021-03-07:
### Bug Description

Not sure if this is failure to understand on my part, or a bug here. With `PRINTER_EVENT_LEDS` whenever the heating is done, I'm seeing the NeoPixel changing to white, but instead of them being set as `R255 U255 B255 P100`, they are set as `R0 U0 B0 W255 P100`

Trying to follow the code, I see this which sets it to white after the heating is done
https://github.com/MarlinFirmware/Marlin/blob/5dc8f5cfabbfcc6803cad1edd3ea469b29b0834a/Marlin/src/feature/leds/printer_event_leds.h#L64

Then this which defines what `set_white()` is
https://github.com/MarlinFirmware/Marlin/blob/5dc8f5cfabbfcc6803cad1edd3ea469b29b0834a/Marlin/src/feature/leds/leds.h#L173

And then it sets the color to `R0 U0 B0 W255`, which isn't valid for a NeoPixel with no separate white LED
https://github.com/MarlinFirmware/Marlin/blob/5dc8f5cfabbfcc6803cad1edd3ea469b29b0834a/Marlin/src/feature/leds/leds.h#L137-L141

If I set my printer to heat up the bed and nozzle to activate this bit of code, then according to Custom Lights in the Lights menu, it's currently set to `R0 U0 B0 W255 P100`. If I adjust the `W` or `P` values from that at all, they go dark. If I power cycle the printer just to make sure no settings are sticking around from the `PRINTER_EVENT_LEDS` code and then send the command `M150 R0 U0 B0 W255 P255`, the NeoPixel goes dark. If I send `M150 R0 U0 B0 W255 P100`, they light up a little under half brightness. If I send an other combination of `W` and `P`, they remain dark. Not sure what is special about these values of `W255 P100`, but from my understanding `W` anything shouldn't do anything.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

BTT SKR 1.4 Turbo updated and compiled 2021-03-07 on `bugfix-2.0.x`
[Marlin SKR 1.4 Turbo Configs 2021-03-07.zip](https://github.com/MarlinFirmware/Marlin/files/6097427/Marlin.SKR.1.4.Turbo.Configs.2021-03-07.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Set bed temp with `M190`
2. Set hotend temp with `M109`
3. Check values of LEDs using the Lights => Custom Lights menu

**Expected behavior:**

The brightness of white used after heating should be set as a value like `R255 U255 B255 P100`, since NeoPixel doesn't have a separate white LED.

Conversely, in the case of NeoPixels without a white LED, `W` values could also be a shortcut of sorts equating to `R255 U255 B255 P<W value>`, and with RGBs without a white LED, the `W` value could be a shortcut for `R<W value> U<W value> B<W value>`. This bit is not strictly related to the issue brought up in the bug report, so I will propose this as a separate feature request.

**Actual behavior:**

After heating, NeoPixels are set to `R0 U0 B0 W255 P100`, which is a bizarre result and confuses the user if they then use the LCD to adjust with Custom Lights setting. Furthermore, `W` anything shouldn't be valid and other values of `W` and `P` are not valid.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

lightmaster, 2021-03-07:
Probably not an ideal solution, but if you change https://github.com/MarlinFirmware/Marlin/blob/5dc8f5cfabbfcc6803cad1edd3ea469b29b0834a/Marlin/src/feature/leds/leds.h#L137-L141

to

```c
#if HAS_WHITE_LED && DISABLED(RGB_LED) && !ENABLED(NEOPIXEL_LED)
  #define LEDColorWhite()         LEDColor(  0,   0,   0, 255)
#else
  #define LEDColorWhite()         LEDColor(255, 255, 255)
#endif
```

then it correctly switches to `R255 U255 B255 W0 P100` after `PRINTER_EVENT_LEDS` finishes.

ellensp, 2021-03-08:
are you aware of this in Marlin/src/feature/leds/leds.cpp
```
  #if ENABLED(NEOPIXEL_LED)
    const uint32_t neocolor = LEDColorWhite() == incol
                            ? neo.Color(NEO_WHITE)
                            : neo.Color(incol.r, incol.g, incol.b, incol.w);
```

lightmaster, 2021-03-08:
If I understand that correctly, that should be setting it to `255,255,255,0` right? But it's not during printer events.

Dtcreation, 2021-03-08:
have you choose  #define NEOPIXEL_TYPE   NEO_GRBW ?

lightmaster, 2021-03-08:
@Dtcreation Nope, using `#define NEOPIXEL_TYPE   NEO_GRB`

X-Ryl669, 2021-03-10:
Are there many users of RGBW led here that set W independantly from R,G,B (something like R255 G0 B0 W255, does it even make sense?) ? 

Because, if we removed the W part in the G-Code command (and internally set the W's value when present to the luminance of the other colors), it'd work flawlessly, IMHO.

X-Ryl669, 2021-03-10:
Said differently, if we linked W as a computation when it's present, instead of having independent value, would it be a big loss ?

X-Ryl669, 2021-03-10:
@lightmaster Did you solve the other Neopixel issue you had ?

lightmaster, 2021-03-11:
@X-Ryl669 No. I ended up upgrading the MKS Robin Nano 1.2 to a BTT SKR 1.4 Turbo with a much bigger screen, built in UART mode for drivers, and none of the issues with NeoPixel like STM32 has.

I still have the Robin Nano 1.2, so whenever you get NeoPixels working on it, I can test to see if it works with the "newer" env.

lightmaster, 2021-05-21:
Don't think this should be closed, as it's still a bug I'm see every time the neopixels change color after heating up.

thisiskeithb, 2021-05-21:
> Don't think this should be closed, as it's still a bug I'm see every time the neopixels change color after heating up.

Please follow the instructions and comment before the 10 days is up next time so the issue remains open:

> This issue has had no activity in the last 60 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 10 days.

lightmaster, 2021-05-21:
Will do next time. Thanks.

ellensp, 2021-05-21:
You might might want to look at the work in progress https://github.com/ellensp/Marlin/tree/bugfix-2.0.x-%23refactor-leds
removing that White component is a right pain..  
HAS_WHITE_LED now works as expected though. 
very little testing has been done so far (ie none other than getting it compiling)

ellensp, 2021-05-22:
@lightmaster can you give this PR a try. Works on my testing. But is a bit of a convoluted mess...

lightmaster, 2021-05-23:
@ellensp Seems to work for my setup.

ellensp, 2021-05-26:
Now merged, closing

closed 2021-05-26 by ellensp