MarlinFirmware/Marlin #21286: [BUG} power supply auto on

theoldfart, 2021-03-08:
Hi Guys, hope I'm in the correct section.
machine:X5S
Marlin v2 bugfix

Upgrading to v2.
On v1.9 the power supply switches on when machine come out of idle. ie Steppers energised. Then shut down when idle>30s

I can't get this to work with v.2. I've been trying various combinations, and pins 66 / 44 /9 . 66/44 goto to a switching board.
v1.9
[quote]
   Select your power supply here. Use 0 if you haven't connected the PS_ON_PIN

   0 = No Power Switch
   1 = ATX
   2 = X-Box 360 203Watts (the blue wire connected to PS_ON and the red wire to VCC)

   :{ 0:'No power switch', 1:'ATX', 2:'X-Box 360' }
*/
#define POWER_SUPPLY 2

#if POWER_SUPPLY > 0
// Enable this option to leave the PSU off at startup.
// Power to steppers and heaters will need to be turned on with M80.
#define PS_DEFAULT_OFF

#define AUTO_POWER_CONTROL        // Enable automatic control of the PS_ON pin
#if ENABLED(AUTO_POWER_CONTROL)
//#define AUTO_POWER_FANS           // Turn on PSU if fans need power
//#define AUTO_POWER_E_FANS
//#define AUTO_POWER_CONTROLLERFAN
#define POWER_TIMEOUT 30
#endif

#endif
[/quote]
v.2
[quote]
/**
   Power Supply Control

   Enable and connect the power supply to the PS_ON_PIN.
   Specify whether the power supply is active HIGH or active LOW.
*/
#define PSU_CONTROL
#define PSU_NAME "Power Supply"

#if ENABLED(PSU_CONTROL)
  #define PSU_ACTIVE_STATE HIGH      // Set 'LOW' for ATX, 'HIGH' for X-Box

  //#define PSU_DEFAULT_OFF         // Keep power off until enabled directly with M80
  //#define PSU_POWERUP_DELAY 250   // (ms) Delay for the PSU to warm up to full power

  //#define PSU_POWERUP_GCODE  "M355 S1"  // G-code to run after power-on (e.g., case light on)
  //#define PSU_POWEROFF_GCODE "M355 S0"  // G-code to run before power-off (e.g., case light off)

  #define AUTO_POWER_CONTROL      // Enable automatic control of the PS_ON pin
    #if ENABLED(AUTO_POWER_CONTROL)
    #define AUTO_POWER_FANS         // Turn on PSU if fans need power
    //#define AUTO_POWER_E_FANS
    //#define AUTO_POWER_CONTROLLERFAN
    //#define AUTO_POWER_CHAMBER_FAN
    //#define AUTO_POWER_E_TEMP        50 // (°C) Turn on PSU if any extruder is over this temperature
    //#define AUTO_POWER_CHAMBER_TEMP  30 // (°C) Turn on PSU if the chamber is over this temperature
    #define POWER_TIMEOUT              30 // (s) Turn off power if the machine is idle for this duration
    //#define POWER_OFF_DELAY          60 // (s) Delay of poweroff after M81 command. Useful to let fans run for extra time.
  #endif
#endif
[/quote]

thanks

ellensp, 2021-03-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

theoldfart, 2021-03-08:
Thanks, I thought that it could be a bug in v.2.

closed 2021-03-08 by thisiskeithb