MarlinFirmware/Marlin #21292: [FR] Non-axis/extruder stepper motor control

radimkarnis, 2021-03-08:
Hello and thank you for all the amazing work that you've done.

### Description
An option to enable direct control of one or multiple stepper drivers. This would allow stepper motors to be used for non-axis/extruder purposes, such as a movable tool dock (see attached concept picture) or a gripper.

My use-case is a simple tool dock with two positions, it doesn't require a precise axis control - just go up and down between two predefined positions. It is held in the upper position with a solenoid and feedback is provided with a limit switch. The only problem is controlling a stepper motor. Generating step/dir signals manually is not a very elegant solution. 

### Feature Workflow

Being able to define one of the stepper drivers as a "utility" motor output and control it would be a welcomed feature for many more use cases.

1. Provide a way to manipulate a "raw" stepper motor (without the need of being described as an extruder - without a temp sensor, ...)

#### Additional Information

[image]

I am sorry if this idea is not very-well described. I was not able to find anything related or get any help on FB/Forum. This feature request originates from my particular use-case, but I can imagine many more. 

If this doesn't make sense, I would appreciate a few pointers in the right direction - I am currently moving the motor with manually generated pulses and I know there has to be a better way

```
for(int x = 0; x < ticks; x++) {
	extDigitalWrite(E7_STEP_PIN, HIGH);
	dwell(1);
	extDigitalWrite(E7_STEP_PIN, LOW);
	dwell(1);
}
```

Thank you!

thisiskeithb, 2021-03-08:
There's an open PR with experimental support for 6 non-extruder axes, but I'm not sure what state things are in: https://github.com/MarlinFirmware/Marlin/pull/19112

radimkarnis, 2021-03-10:
Thank you! This is a great pointer in the right direction. Look like a promising feature for future Marlin versions.
If I wanted to step the motor manually, is there a way to safely wait between the pulses? dwell() and safe_delay() only work with milliseconds, which makes the spinning motion very slow. Are there any similar functions available for microseconds? Or would you solve the problem differently?
Thank you, other than my question - this FR can be closed.

closed 2021-03-16 by radimkarnis