MarlinFirmware/Marlin #21293: [BUG] Password protection doesn't work correctly

Zemistr, 2021-03-09:
### Password protection doesn't work correctly

If you try to enter the Password Settings menu section and you put any password, you get access.

### Configuration Files

Clean `bugfix-2.0.x` with EEPROM_SETTINGS, CR10_STOCKDISPLAY, PASSWORD_FEATURE and PASSWORD_DEFAULT_VALUE = 1234.

### Steps to Reproduce

1. Build marlin with EEPROM_SETTINGS, CR10_STOCKDISPLAY, PASSWORD_FEATURE and PASSWORD_DEFAULT_VALUE = "any".
2. Flash the firmware
3. Initialize EEPROM
4. Store settings
5. Restart the printer
6. Login with your password
7. Open the "Password Settings" menu section
8. Write any password you want.
9. Congrats, now you are HACKER!!! 🍾 

**Expected behavior:**

The same as during login after boot up. 

**Actual behavior:**

Getting access with the wrong password.

ellensp, 2021-03-09:
Steps are not clear.
once you have logged in you should be able to change the password. which is what your steps is doing.

Zemistr, 2021-03-09:
@ellensp I reduced them a little bit.
The steps are really simple. You are able to log in only with the correct password after the boot-up.
If you try to access the password settings menu section the printer will ask you to put the correct password to get access.
The problem is that you can write any password and you get access.

ellensp, 2021-03-09:
no mention of enabling eeprom 
"#warning "PASSWORD_FEATURE settings will be lost on power-off without EEPROM_SETTINGS."

Zemistr, 2021-03-09:
Here is a video https://youtu.be/3EoJFPGrswQ

Zemistr, 2021-03-09:
@ellensp Originally I had the EEPROM in the description, but I just did a test without the EEPROM and the problem is still the same. During login everything works correctly. Marlin validates the password as expected.

ellensp, 2021-03-09:
yes I see what you mean, and the Enter digit  line has garbage at the end too...

Zemistr, 2021-03-09:
Yop, agree. I tried the stable version of Marlin and also the latest bugfix. Both are doing the same.

ellensp, 2021-03-09:
I just found an related bug, still unfixed..
On startup password screen scroll down and select Start Over.
Now wait about 20 seconds, and your in! no password required!

ellensp, 2021-03-09:
I've created a PR to fix these issues.

Zemistr, 2021-03-09:
I tried the login screen and damn... you are right... It's not safe at all... 🤣 
Thanks! Hope the fix will be available before my video about that functionality is released. 😄

thinkyhead, 2021-03-10:
The patch looks good, thanks!

closed 2021-03-10 by thinkyhead