MarlinFirmware/Marlin #21296: [FR] makerbot replicator2  mightyboard rev G and H porting Marling ?

labotecno, 2021-03-09:
### Description
Hello i try to find a porting for mightyboard Rev G and H , i see nothing was made fro this board with mega1280 
and termocouple sensor ADS1118 

### Feature Workflow
i find this one  [https://github.com/rpavlik/Marlin/tree/mightyboard-twox](url)
complation is ok and marlin standing is ok with pronterface 
lcd is not working , and hotend is not heating 

#### Additional Information
May be its so little space of memory but i thinking it will better than sailfish firmware : can t adjust step and velocity or many thing can be marlin 
if someone have this printer , could test this link ? 
thank you

karabas2011, 2021-11-24:
I tried the above Marlin.
The same problems.
No any pronterface input  until I press reset button on board.
Then I see M503 output on monitor. But no any reaction on input commands. No  Ok's
But pronterface stays connected.

caseydunn, 2022-01-02:
There's also a Klipper fork working on the revG/H MightyBoards.

https://github.com/dockterj/klipper

ellensp, 2023-05-07:
https://github.com/rpavlik/Marlin/blob/mightyboard-twox/Marlin/src/pins/mega/pins_MIGHTYBOARD_REVG.h

main thing stopping this is there is no support in standard marlin for ADS1118 ADC, used for Thermocouples

beadon, 2023-05-07:
> https://github.com/rpavlik/Marlin/blob/mightyboard-twox/Marlin/src/pins/mega/pins_MIGHTYBOARD_REVG.h
> 
> main thing stopping this is there is no support in standard marlin for ADS1118 ADC, used for Thermocouples
> 
> also All Mightboards after rev E are closed source. So no schematics, very little available information on the net about them.  

Regarding the ADS1118 - it looks like there's some sample code, and with a quick look, a workable(?) driver here : https://github.com/ReclaimerLabs/ReclaimerLabs_ADS1118

What's preventing support for the ASS1118? ( I may volunteer adding it, but want to understand what I am getting into since I know nothing about the Marlin source code !).

Regarding the board itself, I have a note out to a repair shop that regularly repairs the MightyBoard -- if there's a stream of valuable software that keeps it running, it keeps their businesses going -- so I have asked if they have a schematic.  

Barring the schematic, I can easily get photos of the Gen H board.  Is this helpful in any way ?

Thanks!

edumaker-limited, 2024-09-10:
> Regarding the board itself, I have a note out to a repair shop that regularly repairs the MightyBoard -- if there's a stream of valuable software that keeps it running, it keeps their businesses going -- so I have asked if they have a schematic.
Hi @beadon did you ever get the schematics at all?
I have a large number of Makerbot Replicator 2/2x machines and have been happily running them using the final Makerbot firmware or Sailfish for a long time, but repeatedly looked into porting them to Marlin.
I've built motherboards from scratch and it's 'just an Arduino' so I'm willing to have a go at building a board definition from scratch - a schematic could only help this.

beadon, 2026-01-02:
> Hi [@beadon](https://github.com/beadon) did you ever get the schematics at all? I have a large number of Makerbot Replicator 2/2x machines and have been happily running them using the final Makerbot [...]

Sorry, I was not able to get a schematic.  Given the age of the device and weak performance compared to an inexpensive solution available today I am not pursuing this effort.