MarlinFirmware/Marlin #21297: [BUG] bugfix-2.0.x: "Init KILL, SUICIDE, PSU earlier (#20810)" [ b9ed1395] results in boot loop on LPC1768 platform.

chrisqwertz, 2021-03-09:
### Bug Description

bugfix-2.0.x commit b9ed1395 "Init KILL, SUICIDE, PSU earlier (#20810)" results in boot loop on LPC1768 + RAMPS 1.4 platform.

### Configuration Files

[Conf_files_b9ed1395.zip](https://github.com/MarlinFirmware/Marlin/files/6110496/Conf_files_b9ed1395.zip)

Just to be sure, here's my pins_RAMPS_RE_ARM.h file for my modified RAMPS 1.4 board.

[pins_RAMPS_RE_ARM.zip](https://github.com/MarlinFirmware/Marlin/files/6110505/pins_RAMPS_RE_ARM.zip)

### Steps to Reproduce

Have a LPC1768 + RAMPS 1.4 setup and flash bugfix-2.0.x b9ed1395 

**Expected behavior:**

after a reset the controller fan should start up and stop again for about ~0.4s 3 times in a row before stopping again. The bed would start heating during that short time period. Then the marlin logo should appear in display and the printer should go to idle with all IO pins including bed heater having their respective level set.

**Actual behavior:**

Controller fan spins up and doesn't stop. Heating bed keeps on heating and display stays blank forever.

#### Additional Information

Reverting the commit and going back to 85d61de6 fixes the issue again.

chrisqwertz, 2021-03-09:
Yes sir, in fact, that's why I reverted to a known good and spent quite a while applying, reverting and compiling commits until I narrowed it down to the aforementioned culprit.

Upon closer inspection I might have found a lead:

moving 
```
 #if HAS_TMC220x
    SETUP_RUN(tmc_serial_begin());
 #endif
```
so it sits before the PSU_CONTROL init seems to fix it.

Maybe someone with a similar setup can verify that. 

I vaguely remember having the TMC in the wrong order and breaking the boot sequence a year or so ago.

If I can be of more help, please let me know.

chrisqwertz, 2021-03-09:
I added a proposed fix for the issue: b7f03915cc956c2bb81c17ec86fa196319f62fe7

thisiskeithb, 2021-03-09:
> I added a proposed fix for the issue: [b7f0391](https://github.com/MarlinFirmware/Marlin/commit/b7f03915cc956c2bb81c17ec86fa196319f62fe7)

So this issue isn't resolved?

chrisqwertz, 2021-03-09:
I'm sorry, I didn't word that right: 
What I meant was, on the latest bugfix2.0. commit, I moved the TMC init before the PSU init which fixed the issue.

thisiskeithb, 2021-03-10:
Merged.

closed 2021-03-10 by thisiskeithb