MarlinFirmware/Marlin #21300: [BUG ?] (RUMBA32 + WS2812B )

studiodyne, 2021-03-09:
Specially for @ellensp 
rumba32 + btf lighting led 144/meters RGBW

Folkloric behavior ...
Not supported ?  Lib ? Timing ? Config ? 

Thks

ellensp, 2021-03-10:
If it was for me, you would have followed the template and included config files.

You cannot run 144 neopixels on a 3d printer. setting each led takes a finite time and you cannot take break, you have to set them all at once., this is incompatible with the operation of the firmware, that must do other things at the same time.

closed 2021-03-10 by ellensp

studiodyne, 2021-03-10:
Hi , thk you very much to give your time 
Template ? Config files ?
No ... I just see you have written code for a ws2812 project , and you have enough experience about neopixel

Please give me all , template/config you have  , and how to set ' all at once ' ,  'all at once ' means that only one serial package sent , and auto transmitted to the last chip on the stripe ?

studiodyne, 2021-03-13:
@ellensp  , hi , i'm sorry but i need you again 

I have stopped to use NEOPIXEL , because , timing flow must be perfect , and need a dedicated controler 
I use now 4 pins for RGBW 24v led
But i need PWM on these pins , because now , are just ' ON/OFF'
Pwm on pins are only on dedicated pins or can we switch on any pins we want ?
If pwm pins are hardware only , is it possible to have software pwm for my leds  ' PD12/13/14/15' on RUMBA 32 STMF4

I'm sorry to disturb you , but find these type of information need to search many days in datasheet ,  and to leard marlin very deep

Thks for help

studiodyne, 2021-03-13:
@ellensp 
I need your help , and need a little lesson about PWM
i have replaced neopixel by 4 channel RGBW 24v led , on 4 pins
But i don't have adjustable brightness , just on/off for all leds
Why ?
thks