MarlinFirmware/Marlin #21302: LERDGEK_usb_flash_drive Error: BoardConfig: Board is not defined

Evg33, 2021-03-10:
> Executing task in folder Marlin: C:\Users\xxx\.platformio\penv\Scripts\platformio.exe run --target upload --environment LERDGEK_usb_flash_drive <

Processing LERDGEK_usb_flash_drive (platform: ststm32@~12.0; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Error: BoardConfig: Board is not defined
=================================================== [FAILED] Took 2.40 seconds ===================================================
Environment              Status    Duration
-----------------------  --------  ------------
LERDGEK_usb_flash_drive  FAILED    00:00:02.401
============================================= 1 failed, 0 succeeded in 00:00:02.401 ============================================= 
The terminal process "C:\Users\xxx\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'LERDGEK_usb_flash_drive'" terminated with exit code: 1.

ellensp, 2021-03-10:
[env:LERDGEK_usb_flash_drive]
platform          = ${env:LERDGEK.platform}
extends           = LERDGEK       <-  should this be env:LERDGEK 
platform_packages = ${stm32_flash_drive.platform_packages}
build_flags       = ${stm32_flash_drive.build_flags}

ellensp, 2021-03-10:
confirmed adding env: prefix gets past that error

ellensp, 2021-03-10:
@thisiskeithb this is your 'baby' https://github.com/MarlinFirmware/Marlin/pull/20593 ill leave it to you

thisiskeithb, 2021-03-10:
When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

**This includes attaching your configuration files, regardless of whether you think they are involved.**

For reference, these are the instructions included in the template:

---
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

---

### Bug Description

_Describe the bug in this section._

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

_Describe the steps needed to reproduce the issue._

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

_Describe what you expected to happen here._

**Actual behavior:**

_Describe what actually happens here._

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2021-03-10:
> this is your 'baby' #20593 ill leave it to you

I extended environments using similar ones as an example, so I’m not sure why it doesn’t work. I’ll have to take a look.

ellensp, 2021-03-10:
you have to reference the entire name of block as it was defined

eg
defined as [common_avr8]
extends  = common_avr8

defined as [env:mega2560]
extends   = env:mega2560

so defined as [env:LERDGEK]
extends should be  env:LERDGEK

thisiskeithb, 2021-03-10:
See https://github.com/MarlinFirmware/Marlin/pull/21305

Evg33, 2021-03-10:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6115480/configs.zip)

platformio.ini changed 
```
[env:LERDGEK_usb_flash_drive]
platform          = ${env:LERDGEK.platform}
extends           = env:LERDGEK
platform_packages = ${stm32_flash_drive.platform_packages}
build_flags       = ${stm32_flash_drive.build_flags}
```

```
 C:\cod\Marlin> pio run --environment LERDGEK_usb_flash_drive
Processing LERDGEK_usb_flash_drive (platform: ststm32@~12.0; board: LERDGE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
KeyError: 'usb-host-msc':
  File "C:\users\xxx\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\xxx\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xxx\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\cod\Marlin\buildroot\share\PlatformIO\scripts\copy_marlin_variant_to_framework.py", line 46:
    FRAMEWORK_DIR = platform.get_package_dir(platform_name)
  File "c:\users\xxx\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 33:
    pkg = self.get_package(name)
  File "c:\users\xxx\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "c:\users\xxx\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
=================================================== [FAILED] Took 1.50 seconds ===================================================
Environment              Status    Duration
-----------------------  --------  ------------
LERDGEK_usb_flash_drive  FAILED    00:00:01.498
============================================= 1 failed, 0 succeeded in 00:00:01.498```

thisiskeithb, 2021-05-05:
#21305 was merged. Closing.

closed 2021-05-05 by thisiskeithb