MarlinFirmware/Marlin #21304: SKR E3 Turbo (lpc1769) - compilation failure with binary file transfer enabled

LucienWells, 2021-03-10:
### Bug Description

**Hardware**: BigTreeTech SKR E3 Turbo (LPC1769)

I am trying to compile Marlin-bugfix-2.0.x for the BTT SKR E3 Turbo with binary file transfer enabled.  The firmware compiles fine with binary file transfer commented out in `Configuration_adv.h`, but when the flag is enabled, the firmware fails to compile.

- tested with latest bugfix-2.0.x (downloaded on 10 March 2021)
- I could find no similar bug reports

**Configuration Files**

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/6112413/Configuration.files.zip)

**Steps to Reproduce**

1. Set `default_envs = LPC1769` in `platformio.ini`
2. Set `#define MOTHERBOARD BOARD_BTT_SKR_E3_TURBO` in `Configuration.h`
3. Uncomment `#define BINARY_FILE_TRANSFER` in `Configuration_adv.h`
4. Build the Firmware in VS Code / PIO

**Expected behavior:**

Successful compilation of firmware.

**Actual behavior:**

Compile attempt errors with the following message:

```
Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduino-lpc176x 0.2.6 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/HAL.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/HAL_MinSerial.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/HAL_SPI.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/MarlinSerial.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_flash.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_sdcard.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/eeprom_wired.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/fast_pwm.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/include/digipot_mcp4451_I2C_routines.c.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/include/i2c_util.c.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/main.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/tft/tft_spi.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/tft/xpt2046.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/timers.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/u8g/LCD_I2C_routines.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio/build/LPC1769/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
[... 116 lines cut ...]
      |                                                                                        serial_index_t
Marlin/src/gcode/sd/../../inc/../core/macros.h:477:26: note: in definition of macro 'THIRD'
  477 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin/src/gcode/sd/../../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
  195 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin/src/gcode/sd/../../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
  194 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin/src/gcode/sd/../../inc/../core/macros.h:193:29: note: in expansion of macro '_TERN'
  193 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
      |                             ^~~~~
Marlin/src/gcode/sd/M28_M29.cpp:52:7: note: in expansion of macro 'TERN_'
   52 |       TERN_(HAS_MULTI_SERIAL, card.transfer_port_index = queue.ring_buffer.command_port());
      |       ^~~~~
*** [.pio/build/LPC1769/src/src/gcode/sd/M28_M29.cpp.o] Error 1
======================================================= [FAILED] Took 66.37 seconds =======================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:01:06.368
================================================== 1 failed, 0 succeeded in 00:01:06.368 ==================================================

```

rhapsodyv, 2021-03-10:
Fixed in #21301

thisiskeithb, 2021-03-10:
> Fixed in #21301

Hopefully in https://github.com/MarlinFirmware/Marlin/pull/21306 🙂

LucienWells, 2021-03-11:
> > Fixed in #21301
> 
> Hopefully in #21306 🙂

Can confirm this now works.  

I downloaded the latest bugfix, compiled with `#define BINARY_FILE_TRANSFER` flag set and flashed the board.  

I then used binary file transfer to flash the board again via OctoPrint to ensure it was working.  (It also seems to have resolved an issue where I could not have both the SD card inserted and a USB active for OctoPrint at the same time.)

Thanks all.

closed 2021-03-11 by LucienWells