MarlinFirmware/Marlin #21309: [BUG] Marlin 2.0 filament runout sensor error

rodribike, 2021-03-10:
### Bug Description

 Marlin 2.0 filament runout sensor error
filament is working and status is correct when using pronterface. Also i measured voltage with and without filament and values are ok
when print starts always request to insert filament

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6118478/Configuration.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step] connect printer to pronterface
2. [Second Step] check filament status with M119 -> open
3. [and so on...]insert filament´
4.  M119 --> triggered
5. start print
6. print starts --> filament change
7. turn of filament sensor

**Expected behavior:**

 normal print start with filament

**Actual behavior:** filament is in but printer request to put filament

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rodribike, 2021-03-11:
https://we.tl/t-uePUySU9hN - video with the issue

3dfixer, 2021-03-12:
try in your config.h
comment lines 1223,1227,1228,
uncomment line 1224 #define FIL_RUNOUT_PULLDOWN 
m119 filament should say triggered

works with my bear which looks like yours. skrv1.4 tft35

GoDllkE, 2022-10-18:
I got a similar case, where changing FIL_RUNOUT_STATE to LOW/HIGH also changes the sensor behavior, in other words:
- When I set LOW, the sensor with filament reports open and without filament reports triggered. (Should be opposed)
- When I set HIGH, the sensor with filament reports triggered and without filament reports open. (Should be opposed)

Swapped between FIL_RUNOUT_PULLDOWN and FIL_RUNOUT_PULLUP to see if I got any difference without results. Also tried different cable management in hope it was the case of wrong wiring setup, without results either. (I'm also think i didn't understand properly which those two does)

My setup is an Ender 3 V2, running marlin build environment with 'STM32F103RE_creality' with that Creality motherboard v4.2.7 and latest marlin by the time of this comment, which is 2.1.1:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/9814878/Configurations.zip)

I first noticed this issue in marlin 2.0.9.5 but I decided that my future self would be the one to investigate it further.

As for the sensor, don't know if an image could help, but it's like this:
[image]

Everything else works fine. Any ideas how to make it work?

FlipKeulder, 2024-07-01:
I had the same issue with Marlin 2.1.2.4, for a pultrusion machine (ramps 1.4 and Arduino mega 2560 with ramps full graphic controller screen SD enabled)), needed a run out sensor. Eventually uncommented

#define WATCH_ALL_RUNOUT_SENSORS

in configuration.h which appears to be working. I still get a message with two options when triggered: 
“purge more” and “filament run out: on”. 
It is as if Marlin executes

#define FILAMENT_RUNOUT_SCRIPT
 
when M412 is set to S1, regardless the state of the sensor, triggered or not, High or Low. (I tried all - M119 is good). Anyway, my work around for my situation was to set:

#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 210

#define EMERGENCY_PARSER
 
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT false // Enable the sensor on startup. Override with M412 followed by M500.
  #define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder. Define a FIL_RUNOUT#_PIN for each.

  #define FIL_RUNOUT_STATE      LOW       // Pin state indicating that filament is NOT present.
  #define FIL_RUNOUT_PULLUP               // Use internal pullup for filament runout pins.
  //#define FIL_RUNOUT_PULLDOWN           // Use internal pulldown for filament runout pins.
  #define WATCH_ALL_RUNOUT_SENSORS

#define FILAMENT_RUNOUT_SCRIPT "M104 S0"

This catches the extrude mintemp and stops the pultrusion.
Also added to my gcode M75 and M412 S1

M600 script is not working for me with the “filament run out: on” instead of “Continue” option. 
To recap: Marin did not check sensor state without "#define WATCH_ALL_RUNOUT_SENSORS" enabled, (even with only one sensor) and execute "#define FILAMENT_RUNOUT_SCRIPT" when M412 is set to S1.
Hope this will be of some help.