MarlinFirmware/Marlin #21310: Activate NOZZLE_CLEAN_FEATURE run into "the command line is too long"

cprobst87, 2021-03-10:
If I compile with enabled NOZZLE_CLEAN_FEATURE PlatformIO runs into "the command line is too long":

```
Compiling .pio\build\LPC1769\src\src\sd\SdFile.cpp.o
Compiling .pio\build\LPC1769\src\src\sd\SdVolume.cpp.o
Compiling .pio\build\LPC1769\src\src\sd\cardreader.cpp.o
Linking .pio\build\LPC1769\firmware.elf
Die Befehlszeile ist zu lang.
*** [.pio\build\LPC1769\firmware.elf] Error 1
```

cprobst87, 2021-03-10:
If I short all variables from CLEAN to CN it works, for example: NOZZLE_CN_START_POINT
They are too long...

ellensp, 2021-03-11:
This is a bug in platformio. Not something we can do anything about.
See https://github.com/platformio/platformio-core/issues/3827 for details. and a work around.

closed 2021-03-11 by ellensp