MarlinFirmware/Marlin #21312: PID auto-tuning does not save new data to EEPROM

spammchik, 2021-03-11:
PID auto-tuning in to the "Advanced Settings" menu does not save new data to EEPROM. That is, after performing PID auto-tuning, you see the new PID values, but after restarting the printer, the values return to what they were before the tuning. I would like to, is it possible to make PID saving in EEPROM automatic? Maybe you need to add the functionality of the M303 U command?
Board: SKR V1.3
Marlin v2.0.5.3

ellensp, 2021-03-11:
At the moment you cannot save just these setting you must save all settings. (only mesh data and printer stats are separate) 
To save all setting automatically would be quite bad, Any other temporary setting you might have had at the time would also be saved.

thisiskeithb, 2021-03-11:
> To save all setting automatically would be quite bad

That and you don't always want to overwrite your previous PID values if you're running multiple PID tests on your hotend or bed when troubleshooting or dialing in new hardware.

spammchik, 2021-03-11:
Of course you're right. But the presence of PID Auto-tuning from the printer display in the absence of saving the results of the operation of this function leads to the devaluation of the function itself. You have to write down the obtained values on paper, connect the printer via USB to the computer, and enter the values manually.
If, nevertheless, I want to take risks and save all the EEPROM values after performing PID Autotuning, then how to do it by correcting the firmware. Where to deposit M500?

ellensp, 2021-03-11:
The menu "Store Settings" is available in several menus, depending o what you have enabled.

GhostlyCrowd, 2021-03-15:
@spammchik  there is no need to write anything down, U will load the settings into memory and use them.
U has never saved to eeprom, you still have to save them with M500. It does not save them, the documentation here https://marlinfw.org/docs/gcode/M303.html also mentions saving them in eeprom if desired via M500.  

I have a macro that does the pid auto tune with U and then fires a M500 at the end to save them as well, works fine.

evg-zhabotinsky, 2021-11-06:
Why is this still open?
Just like all the other settings done through LCD, PID autotune results are automatically applied but not saved.
And unless you also click "Store Settings" menu entry or call M500 they are lost upon reboot, as expected.

The only question is whether it needs to somehow be made obvious that all the settings (not just PID) are only saved manually.

closed 2021-11-06 by ellensp