MarlinFirmware/Marlin #21313: [FR] Lerdge-K I²C EEPROM

Evg33, 2021-03-11:
How to make I²C EEPROM work?

Chip FM24CL64 F-RAM 64K (8Kx8)
Datasheet: https://www.cypress.com/file/41656/download

Wiring:
[image]

A0,A1,A2 on gnd. i.e. EEPROM_DEVICE_ADDRESS 0xA0 ?
[image]

```
#define I2C_EEPROM
#define I2C_SDA_PIN                         PG13
#define I2C_SCL_PIN                         PG14
//#define EEPROM_WRITE_DELAY              10
#define EEPROM_DEVICE_ADDRESS           0xA0
//#define MARLIN_EEPROM_SIZE            0x10000 // FM24CL64 F-RAM 64K (8Kx8)
#define MARLIN_EEPROM_SIZE              0x1000  // 4KB for testing
```
In this case M500 gets error "Error writing to EEPROM"

I tried with BL24CXX, but it doesn't compile.
```
#define IIC_BL24CXX_EEPROM  - not compiling
#define IIC_EEPROM_SDA                    PG13
#define IIC_EEPROM_SCL                    PG14
```

What am I doing wrong ?
Need additional code for FM24CL64 ?

Evg33, 2021-03-11:
p.s.: write address +1

GMagician, 2021-03-11:
Not sure, I haven't checked code but maybe address is shifted by code, so you may have to define it as 0x50

Evg33, 2021-03-11:
GMagician: with EEPROM_DEVICE_ADDRESS 0x50 not working :(

```
SENDING:M500
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
....

```

Evg33, 2021-03-11:
Hmmm. on Lerdge X with same i2c chip it working.... PR with pins later....

Evg33, 2021-03-11:
Maybe need changes in Marlin\buildroot\share\PlatformIO\variants\LERDGE\  ?
for example, here are pins for x, but no for k
https://github.com/MarlinFirmware/Marlin/blob/19b94ec3252114ac54bb268bed44679588593729/buildroot/share/PlatformIO/variants/LERDGE/PeripheralPins.c#L106-L128

Evg33, 2021-03-11:
and TIMERS
https://github.com/MarlinFirmware/Marlin/blob/19b94ec3252114ac54bb268bed44679588593729/buildroot/share/PlatformIO/variants/LERDGE/PeripheralPins.c#L132-L208

Evg33, 2021-03-12:
Who can help ?
I can test in HW.

GMagician, 2021-03-12:
I don't know how your board works, but have you seen #21337, try to give it a try

Evg33, 2021-03-12:
I think it won't help. Lerdge-X works with this flash without problems.
But I'll try.

Evg33, 2021-03-13:
with #21337 problem not solved on Lerdge-K :(

```
SENDING:M502
SENDING:M500
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
...
```

dolence, 2021-03-22:
Is there anything I can do to help testing this? I'm trying to move from the awful OFW to Marlin on my Lerdge K too.

dolence, 2021-03-26:
OK, I have this working on K. I 'm not very familiar with git usage and never worked with a project as large as this one, so maybe my approach wasn't the best. I will describe what I did and if anyone else can help me on sending a pull request I would be glad.
The problem is on which pins teh EEPROM is wired on K. While X is using pins connected to a real hardware i2c, Lerdge K board have the chip wired to regular pins and probably do some bitbanging. After talking to Jason Nelson, he suggested me to use softi2c. I'm using SlowSoftI2CMaster together with SlowSoftWire, which is a nice wrapper to Wire library.

This is what I did:
On Marlin\platformio.ini I included this line to self resolve the SlowSoftI2CMaster library dependency.
```
# Feature Dependencies
#
[features]
....
SOFT_I2C_EEPROM         = SlowSoftI2CMaster
                          src_filter=+<src/feature/digipot>
```

Then I downloaded SlowSoftWire.cpp and SlowSoftWire.h from [this github](https://github.com/felias-fogg/SlowSoftWirehttps://github.com/felias-fogg/SlowSoftWire) and manually copied it to Marlin\.pio\libdeps\LERDGEK\SlowSoftI2CMaster.

Made this changes to Marlin\src\pins\stm32f4\pins_LERDGE_K.h
```
#define I2C_EEPROM
#define SOFT_I2C_EEPROM //added this to force the use of software i2c
#define I2C_SCL_PIN                         PG14
#define I2C_SDA_PIN                         PG13
#define MARLIN_EEPROM_SIZE               0x10000
```

On Marlin\src\HAL\shared\eeprom_if_i2c.cpp I made this changes:

```
#if ENABLED(I2C_EEPROM) && DISABLED(SOFT_I2C_EEPROM)
  #include <Wire.h> //use Wire library
#elif ENABLED(SOFT_I2C_EEPROM)
  #include <SlowSoftWire.h> //instead, use this softi2c to Wire wrapper
  SlowSoftWire Wire = SlowSoftWire(I2C_SDA_PIN, I2C_SCL_PIN, true);
#endif
```

Before the changes:
[image]

After the changes, so far working fine!
[image]

Now I'm having some issues with my dual Z connected to Ex-Motor. At this moment I'm writing some details which I will [update in here](https://github.com/MarlinFirmware/Marlin/issues/21414). Any help is appreciated.

GMagician, 2021-03-26:
I see that there is some kind of soft i2c support when IIC_BL24CXX_EEPROM is enabled, maybe this can be used this the changed pinout you discovered

dolence, 2021-03-26:
> I see that there is some kind of soft i2c support when IIC_BL24CXX_EEPROM is enabled, maybe this can be used this the changed pinout you discovered

I will take a look and do some code cleanup this night before I made a PR.

dolence, 2021-04-08:
If anyone wanna try, I just made a change to my fork which should solve this issue dolence/Marlin@c3940a82218e93e27157501ff35528d643a69411 

Also, I would like to know if someone else is having problems with display initialization on K version 2.0.1 board.

sanek88lbl, 2021-04-20:
> If anyone wanna try, I just made a change to my fork which should solve this issue [dolence/Marlin@c3940a8](https://github.com/dolence/Marlin/commit/c3940a82218e93e27157501ff35528d643a69411)
> 
> Also, I would like to know if someone else is having problems with display initialization on K version 2.0.1 board.

Hello. I downloaded your Marlin build, checked the work with epprom - it's ok.
I have Lerdge K v2.0.1 and I did not get the display working on your build. I use my last config - it worked. My config is attached.
However, I couldn't start USB OTG, I couldn't even start the build with usb_otg ga VS Code. Does it work for you? How do I do this?

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/6346992/conf.zip)

dolence, 2021-04-22:
> Hello. I downloaded your Marlin build, checked the work with epprom - it's ok.
> I have Lerdge K v2.0.1 and I did not get the display working on your build. I use my last config - it worked. My config [...]

Your display definition doesn't look right. If it worked, your colors are probably wrong (RGB channels inverted). Try changing TFT_DRIVER ST7796 to TFT_DRIVER LERDGE_ST7796

Are you sure you are using the [bugfix-2.0.x-LerdgeK](https://github.com/dolence/Marlin/tree/bugfix-2.0.x-LerdgeK) branch?
Unless we have different displays, it should be working.

I will try the USB-OTG thing and see if it works. Actually I'm having a problem with dual Z endstops and can't use my printer until it's figured out and have no idea on how to proceed, so I have decided to abandon this firmware and go back to original until someone else help to solve this #21414

sanek88lbl, 2021-04-22:
> Your display definition doesn't look right. If it worked, your colors are probably wrong (RGB channels inverted). Try changing TFT_DRIVER ST7796 to TFT_DRIVER LERDGE_ST7796
Are you sure you are using the bugfix-2.0.x-LerdgeK branch?
Unless we have different displays, it should be working.

Yes, I used the wrong branch, but the problem was the display rotated 180 degrees and the touch screen did not work. I copied from my previous build. Now I used your branch with the correct LERDGE driver: the colors did change, but I liked it more with "TFT_DRIVER ST7796"). I also copied the settings of the touchscreen from my previous config-everything worked, thank you!)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6359450/Configuration.zip)

> I will try the USB-OTG thing and see if it works. Actually I'm having a problem with dual Z endstops and can't use my printer until it's figured out and have no idea on how to proceed, so I have decided to abandon this firmware and go back to original until someone else help to solve this #21414

I'll look at your problem with the second Z-axis motor.
In the meantime, I tried building a LERDGEK_USB_FLASH_DRIVE in VS code:

```
Processing LERDGEK_usb_flash_drive (platform: ststm32@~12.0; board: LERDGE; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
KeyError: 'usb-host-msc':
  File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\user\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\work\3d\firmware\marlin\Marlin-bugfix-2.0.x-LerdgeK\buildroot\share\PlatformIO\scripts\copy_marlin_variant_to_framework.py", line 34:
    FRAMEWORK_DIR = platform.get_package_dir(platform_name)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 33:
    pkg = self.get_package(name)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
================================================= [FAILED] Took 0.84 seconds =================================================

Environment              Status    Duration
-----------------------  --------  ------------
LERDGEK_usb_flash_drive  FAILED    00:00:00.839
============================================ 1 failed, 0 succeeded in 00:00:00.839 ============================================
```

dolence, 2021-04-22:
> Yes, I used the wrong branch, but the problem was the display rotated 180 degrees and the touch screen did not work. I copied from my previous build. Now I used your branch with the correct LERDGE [...]

You should be able to change the display orientation this way:

```
/**
 * TFT Rotation. Set to one of the following values:
 *
 *   TFT_ROTATE_90,  TFT_ROTATE_90_MIRROR_X,  TFT_ROTATE_90_MIRROR_Y,
 *   TFT_ROTATE_180, TFT_ROTATE_180_MIRROR_X, TFT_ROTATE_180_MIRROR_Y,
 *   TFT_ROTATE_270, TFT_ROTATE_270_MIRROR_X, TFT_ROTATE_270_MIRROR_Y,
 *   TFT_MIRROR_X, TFT_MIRROR_Y, TFT_NO_ROTATION
 */
//#define TFT_ROTATION TFT_NO_ROTATION
```

Evg33, 2021-04-22:
> Environment              Status    Duration
> -----------------------  --------  ------------
> LERDGEK_usb_flash_drive  FAILED    00:00:00.839
Use non USB build
https://github.com/MarlinFirmware/Marlin/issues/21302

slowbro, 2021-07-07:
I think this was implemented in #21812.

thisiskeithb, 2021-07-07:
> I think this was implemented in #21812.

Yep and EEPROM pins for the Lerdge X was added in #21322.

closed 2021-07-07 by thisiskeithb