MarlinFirmware/Marlin #21314: No reaction with command G12 P1

cprobst87, 2021-03-11:
The command G12 P0 and P2 are working fine.
Command G12 P1 T2 S1 doesn't work.  No reaction, no moves.

Config with dual extruder:
`#define NOZZLE_CLEAN_FEATURE

#if ENABLED(NOZZLE_CLEAN_FEATURE)
  // Default number of pattern repetitions
  #define NOZZLE_CLEAN_STROKES  12

  // Default number of triangles
  #define NOZZLE_CLEAN_TRIANGLES  3

  // Specify positions for each tool as { { X, Y, Z }, { X, Y, Z } }
  // Dual hotend system may use { {  -20, (Y_BED_SIZE / 2), (Z_MIN_POS + 1) },  {  420, (Y_BED_SIZE / 2), (Z_MIN_POS + 1) }}
  #define NOZZLE_CLEAN_START_POINT { { -120, 0, 0 }, { -120, 0, 0 } }
  #define NOZZLE_CLEAN_END_POINT   { { -100, 0, 0 }, { -100, 0, 0 } }

  // Circular pattern radius
  #define NOZZLE_CLEAN_CIRCLE_RADIUS 6.5
  // Circular pattern circle fragments number
  #define NOZZLE_CLEAN_CIRCLE_FN 10
  // Middle point of circle
  #define NOZZLE_CLEAN_CIRCLE_MIDDLE NOZZLE_CLEAN_END_POINT

  // Move the nozzle to the initial position after cleaning
  #define NOZZLE_CLEAN_GOBACK

  // For a purge/clean station that's always at the gantry height (thus no Z move)
  #define NOZZLE_CLEAN_NO_Z

  // For a purge/clean station mounted on the X axis
  //#define NOZZLE_CLEAN_NO_Y

  // Explicit wipe G-code script applies to a G12 with no arguments.
  //#define WIPE_SEQUENCE_COMMANDS "G1 X-17 Y25 Z10 F4000\nG1 Z1\nM114\nG1 X-17 Y25\nG1 X-17 Y95\nG1 X-17 Y25\nG1 X-17 Y95\nG1 X-17 Y25\nG1 X-17 Y95\nG1 X-17 Y25\nG1 X-17 Y95\nG1 X-17 Y25\nG1 X-17 Y95\nG1 X-17 Y25\nG1 X-17 Y95\nG1 Z15\nM400\nG0 X-10.0 Y-9.0"

#endif`

ellensp, 2021-03-11:
Please attach your configuration files.

cprobst87, 2021-03-11:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6122696/Marlin.zip)
Configuration is in the zip-file.
Thank you!

ellensp, 2021-03-11:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

cprobst87, 2021-03-11:
The same, no reaction. New feature heating before cleaning wipe works, also P0 and P2:
14:07:30.970 : FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 11 2021 14:05:07) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:BIBO2 Touch X EXTRUDER_COUNT:2

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6123195/Configuration.zip)

ellensp, 2021-03-12:
You have
 #define NOZZLE_CLEAN_START_POINT { { -120, 0, 0 }, { -120, 0, 0 } }
 #define NOZZLE_CLEAN_END_POINT   { { -100, 0, 0 }, { -100, 0, 0 } }

where START Y and END Y are both 0. This is not valid

First test is ` if (!diff.x || !diff.y) return;` 
ie if start and end of X are the same or start and end of Y are the same don't continue.

cprobst87, 2021-03-16:
Now it works! Perhaps an echo like "wrong parameter" would be fine ;)

closed 2021-03-16 by cprobst87

ellensp, 2021-03-16:
sanity checks have been added, just waiting for review and inclusion into future marlins.