MarlinFirmware/Marlin #21315: [BUG] define EXTRUDER_OFFSET_Y has not affect

sendel, 2021-03-11:
Test with the `bugfix-2.0.x` branch.

in Configuration.h set 
#define HOTEND_OFFSET_X { 0.0, 34.00 } // (mm) relative X-offset for each nozzle
#define HOTEND_OFFSET_Y { 0.0, 0.6 } 

but only HOTEND_OFFSET_X is work. 
Also HOTEND_OFFSET_Y change to big value, its not affect

ellensp, 2021-03-11:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

hamster65, 2021-03-12:
EXTRUDER_OFFSET is depreciated, use HOTEND_OFFSET.
Are you sure it works for X? It should not....

There is a sanity check, so you should see a corresponding warning when compiling.
from SanityCheck.h:
#elif defined(EXTRUDER_OFFSET_X) || defined(EXTRUDER_OFFSET_Y)
  #error "EXTRUDER_OFFSET_[XY] is deprecated. Use HOTEND_OFFSET_[XY] instead."

sendel, 2021-03-12:
Yes Im write wrong caption. Sorry, in config is HOTEND_OFFSET and X work, but Y not work. Newest bugfix-2.0.x have trouble with MKS TFT, Im think that is in serial port answer. when extruder changed from T1 to T0 and set M109 S250, TFT cant get new value, on marlin_direct_LCD show 250/250, but TFT show 230/250 and waiting for. And Im cant test my file with HOTEND_OFFSET settings in newest firmware. When Im connect PC to serial, may be I will say more about this trouble

sendel, 2021-03-15:
Ok, lastest bugfix-2.0.x  printing with Octoprint. HOTEND_OFFSETs works. Thanx/ With MKS TFT35 (fw 1.0.8) not printing after change tool(hotend), with old version works normal.

closed 2021-03-15 by sendel