MarlinFirmware/Marlin #21323: Touch button over display

luiggies, 2021-03-11:
### Bug Description
I get newest bugfix-2.0.x path and I get default configuration for Tronxy X5SA printer, 
but after firmware updated screen is too small (not used maximum screen area) - see photo 1. 
All touch buttons works correctly, but they located outside printed area.
It's look like that resolution of touch screen is set to 320x240 and resolution of touch sensor set to 480x320.
[image]

Next, I changed screen resolution in Configuration.h file, as bellow:
// TFT Resolution. Enable one of the following options:
//#define TFT_RES_320x240
//#define TFT_RES_480x272
#define TFT_RES_480x320
Result: Screen uses maximum available area - good, but touch button don't working. See photo 2.
[image]

Next I tray change dimension for touch sensor in the file: Marlin>src>lsd>touch>touch_buttons.h
I use 320x240 resolution for touch button configuration, as bellow:
//#define BUTTON_SPACING (((TFT_WIDTH) - (BUTTON_WIDTH * 4)) / 5)
#define BUTTON_SPACING (((320) - (BUTTON_WIDTH * 4)) / 5)
// ...
//#define BUTTON_Y_HI (TFT_HEIGHT) - BUTTON_SPACING
#define BUTTON_Y_HI (240) - BUTTON_SPACING
Result: Invisible touch buttons. Touch screen uses maximum available area, Touch button working correctly but they not printed in touch screen (photo 3)
[image]

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Change display resolution in Configuration file, from #define TFT_RES_320x240 toTFT_RES_480x320
2. Redefine BUTTON_SPACING as: (((320) - (BUTTON_WIDTH * 4)) / 5)
3. Redefine BUTTON_Y_HI as: (240) - BUTTON_SPACING

**Expected behavior:**
Working touch buttons for 480x320 screen resolution.

**Actual behavior:**
For 480x320 screen resolution touch buttons working correctly but they doesn't print on screen.

#### Additional Information
bugfix-2.0.x

[configFiles.zip](https://github.com/MarlinFirmware/Marlin/files/6125994/configFiles.zip)

rhapsodyv, 2021-03-11:
There’s an option specific for tronx x5sa display: TFT_TRONXY_X5SA

did you try it?

luiggies, 2021-03-12:
Yes, I tried it, I disabled TFT_GENERIC and I enabled TFT_TRONXY_X5SA. Effect is the same as photo 2 above.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6128122/Configuration.zip)

[image]

rhapsodyv, 2021-03-12:
So, probably your calibration values are wrong. The X5SA board already ships with right values, so, just comment (or remove) this from your config:
```
  #define TOUCH_CALIBRATION_X -12316
  #define TOUCH_CALIBRATION_Y   8981
  #define TOUCH_OFFSET_X         340
  #define TOUCH_OFFSET_Y         -20
```
(put `//` in front of these lines).

luiggies, 2021-03-12:
I commented these lines but it does not no result. 
Still, problem looks same as in the ticket ( https://github.com/MarlinFirmware/Marlin/issues/21323#issue-829527469 )
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6130442/Configuration.zip)

How can I forced touch screen calibration?

rhapsodyv, 2021-03-12:
send M995 using serial

luiggies, 2021-03-12:
It's helpful. If I understand correctly, in EEPROM was stored incorrect (old) data about touch screen calibration.
Thank you a lot.

rhapsodyv, 2021-03-12:
Exactly! I forgot to ask you to reset eeprom with M502+M500.

Thanks!

closed 2021-03-12 by rhapsodyv

luiggies, 2021-03-12:
I investigate problem deeper and I'm not sure that it is EEPROM issue, because when I remove lines:
  #define TOUCH_CALIBRATION_X -12316
  #define TOUCH_CALIBRATION_Y   8981
  #define TOUCH_OFFSET_X         340
  #define TOUCH_OFFSET_Y         -20
and next I flash firmware, and send command M995 via serial. I calibrate touchscreen once.
(When I turn off printer and next turn on, touch screen calibration is kept.)

But, when I use TOUCH_CALIBRATION and TOUCH_OFFSET as in default configuration file. After every power off -> power on process I must re-calibrate touch screen using M995 cmd.

However, I glad because I know workaround - M995 command

rhapsodyv, 2021-03-12:
The M995 will output on serial the correct values for calibration. You can copy it the configuration file. Did you try it?

luiggies, 2021-03-12:
Result of calibration:

Send: M995
Recv: ok
[...]
Recv: Touch screen calibration completed
Recv: TOUCH_CALIBRATION_X -16929
Recv: TOUCH_CALIBRATION_Y 11211
Recv: TOUCH_OFFSET_X 494
Recv: TOUCH_OFFSET_Y -17
Recv: TOUCH_ORIENTATION TOUCH_LANDSCAPE

For Tronxy default configuration file in GIT repository update will be need?

rhapsodyv, 2021-03-12:
I set a default from the printers I tested. But it’s clear that yours have something different.

You can copy these values to your config file. Then reset eeprom to defaults, and everything should work fine now on.

luiggies, 2021-03-12:
Ok, 
so, thank you for help.