MarlinFirmware/Marlin #21326: [BUG] Error in Binary File Transfer Protocol on Ender 3v2

mriscoc, 2021-03-12:
### Bug Description

The printer stops to respond to g-codes commands after to try a binary file transfer. The file is not closed in the µSD card (0 bytes).

### Configuration Files

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6127899/configuration.zip)

only I enabled BINARY_FILE_TRANSFER

### Steps to Reproduce

I downloaded Bugfix-2.0.x and overwrote the configuration files for Ender3v2 from the configuration repository, changed  default_envs = STM32F103RET6_creality in platformio.ini file.

I enabled BINARY_FILE_TRANSFER and compile.

Try to transfer a .bin file to the printer using two different plugins for Octoprint and it get stuck at 100%:

PROGRESS: 0.00% 5.06KiB/s [5.06KiB/s] Errors: 0
PROGRESS: 0.00% 5.06KiB/s [5.06KiB/s] Errors: 0
PROGRESS: 100.00% 5.06KiB/s [5.06KiB/s] Errors: 0

No more messages. If I press close in the plugin, I can return to Octoprint where the printer is disconnected. If I press Connect, I get a timeout message:

Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor.

Maybe the printer missed the end of binary transmission and not return to the G-code parser. I got a 0 bytes empty file on the µSD. The printer is not stuck, I can use the menus, etc. only does not respond to G-code commands any more.

I made the same test using a bin compiled before of February 25 and the binary file transfer is successful.

thinkyhead, 2021-03-12:
Thanks for the report. Serial code has been under development lately. This may be one one for @X-Ryl669 and @p3p to have a glance at….

X-Ryl669, 2021-03-12:
Looks like a protocol mismatch here, likely the printer is still in binary transfer mode and capturing the serial communication so it does not reach the classic gcode command queue handling anymore. Why and how it got struck...

X-Ryl669, 2021-03-15:
If it fixes the error, I think we should increase it by default to 128 bytes (from 64 bytes) since it does not make sense anymore to have such a low buffer.

p3p, 2021-03-15:
The size of the payload buffer shouldn't have an effect on whether a transfer completes, smaller buffer would just have a slower transfer rate, (more packets to ack, more latency), the default 64byte buffer is plenty for uart and on error there is less data to resend.

NezSez, 2021-03-15:
Packet fragmentation due to buffer size shouldn't be an issue as p3p notes, unless using very tiny/huge buffer sizes, or very slow processors, which is not the case here.

If going from 64 to 128bytes on the RX buff alters the result, is there anything down/upstream in the computations that might have to recompute (think packet fragmentation or resulting denial-of-service/starvation/infinite wait loop scenarios here) that would hang on say a last 64byte buffer bucket but NOT hang with 128/256byte being the only difference? That would be very odd since closing/destroying/flushing the last buff bucket would use the same code regardless; unless some function never feeds/receives the proper ACK/NAK/closing sequence.

Maybe some function does not restore some default state on a per mcu/cpu/machine setting/port basis, like resetting G0/G1 moves for Marlin compatible gcode, releasing the shared/unshared serial port for use, etc, when the last buffer read/close/flush or the close of the write to µSD returns...

but again if it works with 128bytes being the only change that would be bizarre.

Maybe the size of the file transferred is hitting some other limit, like what happens when trying to transfer a large file over ESP01/s wifi?

ldursw, 2021-03-15:
I've just tested again to make sure and I can reliably reproduce the issue when `RX_BUFFER_SIZE` is set to 64 or commented out (defaults to 64).

The transfer does not start or fails a few moments after starting it.
```
Starting upload to OctoPrint server
UPLOAD PROGRESS: 100
Upload to server done
Starting transfer to Marlin
Starting transfer of marlinbft/Robin_e3.bin to Robin_e3.bin on remote
parsererror
pySerial Version: 3.5
Connecting: Switching Marlin to binary protocol...
Connection synced [0], binary protocol version 0.1.0, 96 byte payload buffer
File Transfer version: 0.1.0, compression: heatshrink
Opened file: Robin_e3.bin
```

It got stuck there, a few seconds later the SD card icon in the LCD disappeared but the file name remained. Trying to connect again results in a timeout and doesn't connect again until power cycling the printer. The LCD functions are still operational.
```
Connecting to port /dev/ttyUSB0, baudrate 921600
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x7f62077128, open=True>(port='/dev/ttyUSB0', baudrate=921600, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Send: N0 M110 N0*125
Connection closed, closing down monitor
Changing monitoring state from "Connecting" to "Offline"
```

The file is created in the SD card with size 0 bytes. Power cycling with this file present was enough for the bootloader to try to update the firmware and render the printer unbootable. I had to flash the firmware again.

If `RX_BUFFER_SIZE` is set to 128 everything works perfectly. The file is transferred to the SD card with no corruption, the printer reboots, flashes the firmware from the SD card and reconnects successfully to OctoPrint.

Here is the [upload log when set to 128](https://github.com/MarlinFirmware/Marlin/files/6144807/bft128.txt) And the [firmware image](https://github.com/MarlinFirmware/Marlin/files/6144823/Robin_e3.zip) I used. The board is an MKS Robin E3. STM32F103RCT6 with a CH340G for the USB serial.

The small buffer also caused #21170. As far as I can tell libmaple is corrupting the serial buffer and Marlin has no way to tell if the command was corrupted or not. The gcode parser can detect it to some degree and ask the host to resend. The corruption may have exposed an edge case in the BFT code that makes it wait forever.

rhapsodyv, 2021-03-15:
For me, it only works if I set `-d 50` packet size to less than 53.

mriscoc, 2021-03-15:
I increase the RX buffer with:  #define RX_BUFFER_SIZE 128 but the BUG persist.

rhapsodyv, 2021-03-16:
> I increased the RX buffer with: #define RX_BUFFER_SIZE 128 but the BUG persist.

can you try with `-d 50` option to the `transfer.py`?

p3p, 2021-03-16:
For @mriscoc the same transfer was working in February, nothing has changed with the BFT protocol in over a year so I'm assuming something else has changed in the serial pipeline caused by the serial refactor. There are other issues with the protocol, (including issue 64byte intermediate buffer @rhapsodyv is talking about) the version in Marlin was a proof of concept and they are being addressed in the finished version.

p3p, 2021-03-16:
It's interesting how its failing, and the 0 byte file on the sd card even after showing some data transferred, can you try with a dos compatible filename such as `A.BIN` to see if that makes a difference.

mriscoc, 2021-03-16:
> For @mriscoc the same transfer was working in February, nothing has changed with the BFT protocol in over a year so I'm assuming something else has changed in the serial pipeline caused by the serial refactor. There are other issues with the protocol, (including issue with a 64byte intermediate buffer on some STM boards @rhapsodyv is talking about) the version in Marlin was a proof of concept and they are being addressed in the finished version.

I will try ir later.

mriscoc, 2021-03-16:
> It's interesting how its failing, and the 0 byte file on the sd card even after showing some data transferred, can you try with a dos compatible filename such as `A.BIN` to see if that makes a difference.

I tried always with 8.3 names (F1.BIN) also with small gcodes, the same problem.

mriscoc, 2021-03-16:
> > I increased the RX buffer with: #define RX_BUFFER_SIZE 128 but the BUG persist.
> 
> can you try with `-d 50` option to the `transfer.py`?

I will try it later.

NezSez, 2021-03-16:
@ldursw do you know which code issued line 6 "parsererror"?

> Starting upload to OctoPrint server
> UPLOAD PROGRESS: 100
> Upload to server done
> Starting transfer to Marlin
> Starting transfer of marlinbft/Robin_e3.bin to Robin_e3.bin on remote
> parsererror

Perhaps after successfully uploading from computer to OctoPrint, the plugin on OctoPrint which starts the transfer to the SD on the printer isn't properly reading the uploaded file? Maybe the transfer didn't properly EOL/EOF the file with C style or a non-visual char sequence got embedded or something similar which is screwing with the parse and/or closing the write?

mriscoc, 2021-03-16:
> > I increased the RX buffer with: #define RX_BUFFER_SIZE 128 but the BUG persist.
> 
> can you try with `-d 50` option to the `transfer.py`?

I did use: 
```
python ./transfer.py F1.bin -b 250000 -p /dev/ttyUSB0 -d 50
```
with default RX_BUFFER_SIZE and the transfer **was successful**!

ldursw, 2021-03-16:
Testing with a Creality 4.2.2 board using the provided configuration I managed to reproduce the issue.

- Increasing the buffer does not help.
- Transfering from CLI always works (with or without `-d 50`)
- Transfering from OctoPrint breaks after ec42be3

[bftbad.txt](https://github.com/MarlinFirmware/Marlin/files/6149329/bftbad.txt) at 5ffa6f3
[bftgood.txt](https://github.com/MarlinFirmware/Marlin/files/6149394/bftgood.txt) at d5b0662

@mriscoc Your command has a baudrate of 250000 but the configuration in the first post is 115200. Did you change something since opening the issue?

All the tests above were done at 115200 with default RX buffer.

@NezSez `parsererror` is present even if the transfer succeeds. I'm not sure what that means.

mriscoc, 2021-03-16:
> Testing with a Creality 4.2.2 board using the provided configuration I managed to reproduce the issue.
> 
> * Increasing the buffer does not help.
> * Transfering from CLI always works (with or without `-d 50`)
> * Transfering from OctoPrint breaks after [ec42be3](https://github.com/MarlinFirmware/Marlin/commit/ec42be346d5c0d072feb8a1b63ef2fdbd6dc1e98)
> 
> Differently from the other board this one transfers the file and then gets stuck at 100%. The file in the SD card is 0 bytes.
> 
> [bftbad.txt](https://github.com/MarlinFirmware/Marlin/files/6149329/bftbad.txt) at [5ffa6f3](https://github.com/MarlinFirmware/Marlin/commit/5ffa6f3331569c8fa0266c1395d7a6a8cf5515f9)
> [bftgood.txt](https://github.com/MarlinFirmware/Marlin/files/6149394/bftgood.txt) at [d5b0662](https://github.com/MarlinFirmware/Marlin/commit/d5b06624fb2b14b57a9621d7b86caf9f1c5dbfb9)
> 
> @mriscoc Your command has a baudrate of 250000 but the configuration in the first post is 115200. Did you change something since opening the issue?
> 
> All the tests above were done at 115200 with default RX buffer.
> 
> @NezSez `parsererror` is present even if the transfer succeeds. I'm not sure what that means.

I want to confirm, with CLI always works.

My actual installed firmware uses 250000 bauds, but on my initial post I used 115200. With both speeds I have the same results.

mriscoc, 2021-03-16:
Some more notes: 
If I try to transfer using plugins and the printer stops to respond, I can restore the communication using the transfer.py, the CLI tries to start, waits a few seconds and after some errors the transfer succeeds. Then I can to connect to serial and the printer respond to g-codes normally.

ldursw, 2021-03-18:
@mriscoc Please test if this patch fixes the issue for you.

```patch
diff --git a/Marlin/src/feature/binary_stream.h b/Marlin/src/feature/binary_stream.h
index 9f3702b5cec2a61c4578edca4bb0301efd07c288..4d374242f3d6b32abb2c47d4ef09b0b70d484305 100644
--- a/Marlin/src/feature/binary_stream.h
+++ b/Marlin/src/feature/binary_stream.h
@@ -39,7 +39,7 @@ inline int bs_read_serial(const serial_index_t index) {
 
 #if ENABLED(BINARY_STREAM_COMPRESSION)
   static heatshrink_decoder hsd;
-  static uint8_t decode_buffer[512] = {};
+  static uint8_t decode_buffer[MAX_CMD_SIZE] = {};
 #endif
 
 class SDFileTransferProtocol  {

```

The difference between the OctoPrint plugin and CLI is that the plugin has compression always enabled. The CLI has compression disabled by default and can be enabled by passing `-c`. When compression is enabled the issue occurs in CLI as well.

When compression is disabled the SD card is written in `MAX_CMD_SIZE` chunks but when it is enabled the SD card is written in 512-byte chunks. `CardReader::write` fails with `-1` when writing the large chunk. This patch makes the card operations work with the same size when compression is enabled or disabled.

p3p, 2021-03-18:
Thanks for looking deeper into this, although I do find it odd that writing to the sdcard is failing for 512B chunks for multiple people, and why it has just started failing now. 512byte writes to the sdcard should be fine for all cards.

mriscoc, 2021-03-19:
> @mriscoc Please test if this patch fixes the issue for you.
> 
> ```diff
> diff --git a/Marlin/src/feature/binary_stream.h b/Marlin/src/feature/binary_stream.h
> index 9f3702b5cec2a61c4578edca4bb0301efd07c288..4d374242f3d6b32abb2c47d4ef09b0b70d484305 100644
> --- a/Marlin/src/feature/binary_stream.h
> +++ b/Marlin/src/feature/binary_stream.h
> @@ -39,7 +39,7 @@ inline int bs_read_serial(const serial_index_t index) {
>  
>  #if ENABLED(BINARY_STREAM_COMPRESSION)
>    static heatshrink_decoder hsd;
> -  static uint8_t decode_buffer[512] = {};
> +  static uint8_t decode_buffer[MAX_CMD_SIZE] = {};
>  #endif
>  
>  class SDFileTransferProtocol  {
> ```
> 
> The difference between the OctoPrint plugin and CLI is that the plugin has compression always enabled. The CLI has compression disabled by default and can be enabled by passing `-c`. When compression is enabled the issue occurs in CLI as well.
> 
> When compression is disabled the SD card is written in `MAX_CMD_SIZE` chunks but when it is enabled the SD card is written in 512-byte chunks. `CardReader::write` fails with `-1` when writing the large chunk. This patch makes the card operations work with the same size when compression is enabled or disabled.

Thank you, I applied your patch and all is ok now.

X-Ryl669, 2021-03-19:
Can you try with something less than 512 (like 256) instead of MAX_CMD_SIZE (that's likely 96 ?). I'm wondering if there is a relation between the decode buffer size or if it's related to the decompressor status (it's more likely able to decompress in one shot if the decode_buffer is larger than twice the input size, and this might be the bug). 

When using MAX_CMD_SIZE, you restrict the decompressor to stop before it has completely decompressed the input buffer and this could follow another branch of the code that's working correctly.

mriscoc, 2021-03-21:
> Can you try with something less than 512 (like 256) instead of MAX_CMD_SIZE (that's likely 96 ?). I'm wondering if there is a relation between the decode buffer size or if it's related to the [...]

I did use 256 and was also ok.

ldursw, 2021-03-25:
@mriscoc Please test the latest bugfix. A better fix has been merged.

mriscoc, 2021-03-25:
This bug was fixed with the PR: "[Fix buffer alignment for SDIO #21396](https://github.com/MarlinFirmware/Marlin/pull/21396)"

closed 2021-03-25 by mriscoc