MarlinFirmware/Marlin #21330: [FR] Small (<32Kb) I2C EEPROM doesn't work

jafal99, 2021-03-12:
### Description
Current HAL code has support  I2C EEPROM but if connected chip has size less then 32Kb (4KB) it can't be accessed. This is due the fact that smaller chips use only one byte addressing. I have prepared small fix for HAL code to allow using small and large I2C EEPROMs. Detection rely on MARLIN_EEPROM_SIZE

### Patch
[small_i2c_eeprom.zip](https://github.com/MarlinFirmware/Marlin/files/6129018/small_i2c_eeprom.zip)

```
diff --git a/Marlin/src/HAL/shared/eeprom_if_i2c.cpp b/Marlin/src/HAL/shared/eeprom_if_i2c.cpp
index cc22bf7d98..98e71dedb4 100644
--- a/Marlin/src/HAL/shared/eeprom_if_i2c.cpp
+++ b/Marlin/src/HAL/shared/eeprom_if_i2c.cpp
@@ -45,6 +45,24 @@ void eeprom_init() { Wire.begin(); }

 static constexpr uint8_t eeprom_device_address = I2C_ADDRESS(EEPROM_DEVICE_ADDRESS);

+void _beginTransmission(const uint16_t memoryAddress)
+{
+  if (MARLIN_EEPROM_SIZE > 0x4000)     //use two byte addressing for eeproms >16kB
+  {
+     Wire.beginTransmission(eeprom_device_address);
+    // Address High Byte
+    Wire.write((memoryAddress >> 8));
+  }
+  else
+  {
+    uint8_t addr = eeprom_device_address | (byte)((memoryAddress >> 8) & 0x07);
+    Wire.beginTransmission(addr);
+  }
+
+   // Address Low Byte (or only byte for chips 16K or smaller that only have one-word addresses)
+   Wire.write((memoryAddress & 0xFF));
+}
+
 // ------------------------
 // Public functions
 // ------------------------
@@ -52,9 +70,7 @@ static constexpr uint8_t eeprom_device_address = I2C_ADDRESS(EEPROM_DEVICE_ADDRE
 void eeprom_write_byte(uint8_t *pos, unsigned char value) {
   const unsigned eeprom_address = (unsigned)pos;

-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write(int(eeprom_address >> 8));   // MSB
-  Wire.write(int(eeprom_address & 0xFF)); // LSB
+  _beginTransmission(eeprom_address);
   Wire.write(value);
   Wire.endTransmission();

@@ -66,11 +82,14 @@ void eeprom_write_byte(uint8_t *pos, unsigned char value) {
 uint8_t eeprom_read_byte(uint8_t *pos) {
   const unsigned eeprom_address = (unsigned)pos;

-  Wire.beginTransmission(eeprom_device_address);
-  Wire.write(int(eeprom_address >> 8));   // MSB
-  Wire.write(int(eeprom_address & 0xFF)); // LSB
+  _beginTransmission(eeprom_address);
   Wire.endTransmission();
-  Wire.requestFrom(eeprom_device_address, (byte)1);
+
+  int addr = eeprom_device_address;
+  if(MARLIN_EEPROM_SIZE <= 0x4000){ //for eeproms <=16kB lower adres bits ar used for 2kB page address
+     addr = eeprom_device_address | (byte)((eeprom_address >> 8) & 0x07);
+  }
+  Wire.requestFrom(addr, (byte)1);
   return Wire.available() ? Wire.read() : 0xFF;
 }
```

thinkyhead, 2021-03-12:
Thanks! I've made this into a PR on your behalf.

closed 2021-03-12 by thinkyhead

thinkyhead, 2021-03-12:
I see that we have a separate EEPROM implementation for BL24Cxx that accounts for the address different, so maybe that can be unified with this more general i2c-based EEPROM code at some point in the future.

jafal99, 2021-03-14:
I did quick look on BL24CXX:

- the BL24CXX chips are just regular I2C EEPROMs
- looks that the BL24CXX library is used only by STM32F1 HAL and it is wrapped into eeprom_if 
- Ender DWIN controller directly uses BL24CXX library for reading/writing one byte - this can be replaced eeprom_if call
- the library uses its own implementation of software I2C which allows to specify SCL and SDA pins.  I think the story behind this is like: The STM32F1 (and STM32F4 also) has two hardware I2C controllers: first on pins PB6,7 ans second on pins PB10,11. In addition  first controller can be remapped to PB8,9.  Maple Arduino core used by Marlin creates global `Wire` instance for first I2C (on pins PB6,7) and most libraries uses this global instance. To use the second I2C new instance must be created by calling `TwoWire Wire2(2);` and then use Wire2 instead of Wire in code.  According to pins on CREALITY boards BL24CXX EEPROM connected to PA11,12 not to hardware SPI. Maple core also provides `SoftWire` - software implementation for I2C on any pins. This can be used instead of implementation in this BL24CXX lib. 

The questions are: 

1. how to replace global instance of Wire with new one, 
2.  Shall be separate instance of I2C used only for EEPROM? (I dont't think so)

The possibility of specifying which I2C controller (harware1,hardware1alt,hardware2, software) shall be used can be useful in general, not only for Ender. To use second controller I need to alter core library

thinkyhead, 2021-03-18:
Apparently this change breaks EEPROM functionality for many boards. I will revert the changes from #21337 until we can figure out where the error lies.