MarlinFirmware/Marlin #21339: [BUG] Not Z homing using default configuration for Tronxy X5SA

luiggies, 2021-03-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I get newest bugfix.2.0.x and newest configuration file for Tronxy X5SA.
I flashed printer and I tray homing it but Z axis not homing.

When I call G28 command (or use Auto home from printer's menu):
1. Bed goes down 10 mm - correct
2. homing X axis - correct
3. homing Y axis - correct 
4. extruder goes to center point of the bed (XY move) - correct
5.*should homing Z axis but bed does not move - Z axis not homing - NOT correct. 
6. Bed goes down 10 mm again - correct 

* step 5 is very fast. Move from step 4 to 6 is very fast. Step 5 is not observable.  

Tronxy X5SA ueses to homing and bed probe the same induction sensor, so
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and USE_PROBE_FOR_Z_HOMING are defined

### Expected behavior

Z homing correctly

### Actual behavior

1. Distance between bed and extruder increase to 10 mm
2. Homing XY correctly 
3. Homing Z - not effect
4. Distance between bed and extruder increase to 10 mm again.

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Tronxy X5SA

### Electronics

BOARD_CHITU3D_V5

rhapsodyv, 2021-03-12:
Change to BOARD_CHITU3D_V6

luiggies, 2021-03-12:
On the BOARD_CHITU3D_V6 during homing procedure:
Bed moves up (good), but soft doesn't listen inductive Z endstop - bed collision with extruder.

[image]

BTW: earlier issue with touch screen doesn't appear

rhapsodyv, 2021-03-12:
This is exactly the same board I have. You need to use V5 board + this env `chitu_v5_gpio_init`

luiggies, 2021-03-13:
Not good result. I set environment:

> [platformio]
> src_dir      = Marlin
> boards_dir   = buildroot/share/PlatformIO/boards
> default_envs = chitu_v5_gpio_init
> include_dir  = Marlin

and board defined as BOARD_CHITU3D_V5
but result is the same as main ticket description: https://github.com/MarlinFirmware/Marlin/issues/21339#issue-830578612

rhapsodyv, 2021-03-13:
Can you try my tronxy branch?

https://github.com/rhapsodyv/Marlin

Because what you described is either a issue with the board version, or a issue with gpio init env, or a hardware issue.

luiggies, 2021-03-16:
I tried it and effect is still the same, no correctly Z-homing.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6151489/Configuration.zip)
On stock firmware Tronxy homing correctly.
any more advice?

Dark-Lancer, 2021-03-21:
I had a similar problem on mz xy-2 pro.
I checked the endstop states with pronterface M119. After power on my z-sensor was at High until i triggered it 1 time manually.
I uncommented the following lines in configuration.h:

1030   #define ENDSTOPPULLUP_ZMIN
1031   #define ENDSTOPPULLUP_ZMIN_PROBE

then it works for me.

luiggies, 2021-03-24:
Thank you for help but doesn't work - probably I must to change the mainboard...

closed 2021-03-30 by luiggies

rhapsodyv, 2021-03-30:
@luiggies did it work for you?

luiggies, 2021-04-07:
@rhapsodyv, not for the stock mainboard, I going to change it. I bought SKR v1.3, but I don't change it, yet.

luiggies, 2021-04-08:
@rhapsodyv about tronxy mainboard: I noticed that, after starting the printer, Z homing does not work correctly, 
but if i activate Z probing endstop (example I touch this using some piece of metals) before homing, homing procedure works correctly.

Dark-Lancer, 2021-04-08:
@luiggies This is what i mentioned. Check the Endstoppullups in the marlin config. this worked for me.

luiggies, 2021-04-08:
@Dark-Lancer I tried your solution but it does not working for me.

Dark-Lancer, 2021-04-08:
Hmmm, okay. I have the V6 board.
Just as an idea, have you tried to use the pulldown settings in marlin? Maybe the boards have other internal routings.
In a forum, someone had the similar error, but also a v6 board :/ [https://drucktipps3d.de/forum/topic/tutorial-tronxy-x5sa-von-chitu-auf-marlin/?part=14#postid-135991](url) In Posting #138 he has written his found solution.

Otherwise i have no ideas left.
Wish u luck to get it running.

Dark-Lancer, 2021-05-08:
@luiggies I found a second changed in my files which i haven't remembered.

I had to change the assigned pin on my v6 board, but i found no information if it's the same on the v5 board.

In the File: buildroot\share\PlatformIO\variants\marlin_CHITU_F103\wirish\board_setup.cpp i changed the GPIO-PIN for the Zmin/Sensor in Line 103
Original from Rhapsody:
`GPIOA->regs->BSRR = (1U << PA14);`
My Change:
`GPIOA->regs->BSRR = (1U << PG9);`