MarlinFirmware/Marlin #21342: [BUG] lcd_pause_show_message was not declared in this scope when enabling ADVANCED_PAUSE_FEATURE

MarcelRobitaille, 2021-03-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I seem to be having the same issue as #17672.

When I enable ADVANCED_PAUSE_FEATURE, which I need for my filament runout sensor, I get the error message:
```
In file included from Marlin/src/lcd/../inc/../core/boards.h:24:0,
                 from Marlin/src/lcd/../inc/MarlinConfigPre.h:37,
                 from Marlin/src/lcd/../inc/MarlinConfig.h:28,
                 from Marlin/src/lcd/ultralcd.cpp:23:
Marlin/src/lcd/ultralcd.cpp: In static member function 'static void MarlinUI::pause_print()':
Marlin/src/lcd/ultralcd.cpp:1514:28: error: 'lcd_pause_show_message' was not declared in this scope
       TERN_(HAS_WIRED_LCD, lcd_pause_show_message(PAUSE_MESSAGE_PARKING, PAUSE_MODE_PAUSE_PRINT)); // Show message immediately to let user know about pause in progress
                            ^
Marlin/src/lcd/../inc/../core/macros.h:414:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin/src/lcd/../inc/../core/macros.h:195:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin/src/lcd/../inc/../core/macros.h:194:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/lcd/../inc/../core/macros.h:193:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
                             ^~~~~
Marlin/src/lcd/ultralcd.cpp:1514:7: note: in expansion of macro 'TERN_'
       TERN_(HAS_WIRED_LCD, lcd_pause_show_message(PAUSE_MESSAGE_PARKING, PAUSE_MODE_PAUSE_PRINT)); // Show message immediately to let user know about pause in progress
       ^~~~~
```

If I comment out `#define ADVANCED_PAUSE_FEATURE` in `Configuration_adv.h`, it compiles succesfully. I am using a mega with a ramps shield. I used to use filament runout just fine on the same hardware in Marlin 1.x.

### Bug Timeline

Earliest report in Apr 23, 2020

### Expected behavior

Compiles successfully

### Actual behavior

Error

### Steps to Reproduce

1. Checkout bugfix-2.0.x
1. Copy default delta configuration files into `./Marlin`
1. Enable `ULTRA_LCD`
1. Enable `NOZZLE_PARK_FEATURE`
1. Enable `ADVANCED_PAUSE_FEATURE`
1. `pio run`

### Version of Marlin Firmware

Reproduced on latest bugfix (e9f7003b70331533c5d782bb3eb77f106ef89137)

### Printer model

3KU Delta

### Electronics

mega2560 with ramps shield

### Add-ons

LCD?

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

[config_bug_report.zip](https://github.com/MarlinFirmware/Marlin/files/6134144/config_bug_report.zip)

ellensp, 2021-03-13:
Cannot replicate this error. Usingcurrent bugfix and current bugfix configuration example generic delta with enabling the 3 enables.
I do how ever get a 
Marlin/src/HAL/AVR/../../inc/SanityCheck.h:874:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."

MarcelRobitaille, 2021-03-13:
Isn't `ULTRA_LCD` an LCD controller? I did enable `EMERGENCY_PARSER` too to fix a similar error. Sorry, I forgot to mention it.

MarcelRobitaille, 2021-03-13:
@ellensp Your hint led me to check my lcd controller again. I have to enable `REPRAP_DISCOUNT_SMART_CONTROLLER`.

closed 2021-03-13 by MarcelRobitaille