MarlinFirmware/Marlin #21345: (FR) User selectable probing speeds

rq3, 2021-03-13:
### Describe the feature you want

I have a delta printer with an under bed piezo probe, which requires a very slow (10 mm/sec) nozzle contact speed. Ideally, I'd like to see the nozzle move quickly from the home position to within 5 millimeters of the bed during M48, delta calibration, and automated bed leveling.

I've tried every possible thing I can think of, but the nozzles moves from home (at the top of the delta, to where it would normally request a deploy or stow, (which I do not need and have turned off). This is about 350 millimeters above the bed.

It then proceeds at 10 mm/sec until the nozzle contacts the bed, and proceeds to do as was requested. This is painful, since it obviously takes an unnecessary 35 seconds just to get near the bed.

It would be great if there was a way to tell the printer to get within X millimeters of the bed before slowing to the probing speed.

rq3, 2021-03-14:
I just discovered that this does work as I expected if multiple probing is disabled.

thisiskeithb, 2022-01-20:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/18843

closed 2022-01-20 by thisiskeithb