MarlinFirmware/Marlin #21351: [BUG] Second print after the first without resetting crashes head

kpishere, 2021-03-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After reset and first print all is good.  Using 3-point auto bed levelling (does this with other bed levelling schemes though).  However, if one immediately does a second print without resetting firmware, the head crashes into build plate just after z-probing.  This configuration is using an induction sensor.

There appears to be some kind of global variable that is not initialized at start of print and some adjustment values are being added twice? 

Note that all is configured such that no baby-stepping is required.

I'm using Cura as a slicer.  Here is my opening and closing parts for the slicing files. 

Opening: 

G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
G28 ; home all 
G29 ; mesh bed leveling
M300 S660 P500 ; bell
G92 E0.0 ; reset extruder distance position
G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position

Closing: 

M104 S0 ; turn off extruder
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X0 Y210; home X axis and push Y forward
M84 ; disable motors

Here is an eeprom snapshot : 

>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E398.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 P22.52 I1.91 D66.42
echo:  M304 P52.54 I9.72 D189.38
echo:; LCD Contrast:
echo:  M250 C190
echo:; Z-Probe Offset (mm):
echo:  M851 X-46.00 Y2.00 Z-0.28
echo:; Stepper driver current:
echo:  M906 X650 Y650 Z800
echo:  M906 T0 E550
echo:; StallGuard threshold:
echo:  M914 X70 Y70 Z50
echo:; Driver stepping mode:
echo:  M569 S1 X Y
echo:  M569 S1 T0 E
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U100.00

And the config I'm using are here in the iTopie400+ branch :  

[gitlink](https://github.com/kpishere/Marlin/tree/itopie400+)

### Bug Timeline

Has been there for a few months.

### Expected behavior

I'd expect that I can make multiple prints without resetting the firmware first.

### Actual behavior

Head crashes after the second print from reset or power on.

### Steps to Reproduce

Turn on, print any gcode file.  Then print a second one without reset.  See the gcode start in the description, always doing G28 then G29 before each print.

### Version of Marlin Firmware

Latest branch on bugfix-2.0.x.  Referenced repo in description shows exact commit of this branch.

### Printer model

Custom RepRap iTopie

### Electronics

SMART RAMPS and Arduino MEGA2560 clone

### Add-ons

Induction z-probe.  TMC2209 drivers UART mode, sensorless homing at Z-MAX

### Your Slicer

Cura

### Host Software

SD Card (headless)

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/6137222/Archive.zip)

kpishere, 2021-03-20:
Since posting this, I've switched to manual bed leveling and updated the start code of cura to be as follows :  

'''
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
G28 ; home all 
; G29 ; mesh bed leveling
M420 S1 ; use stored mesh bed leveling
M300 S660 P500 ; bell
G92 E0.0 ; reset extruder distance position
G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position
'''

Printed a part the other day, 3-times.  Turning off printer and starting from cold.  Each time, I must increase the baby-stepping by about -0.97 mm.  Yes, head crashing as it comes down, I compensate by increasing baby-stepping as it comes down.  It looks like the offset is being stored in EEPROM but I don't see anything off kilter with the M503 command.  Here is output after those few prints.

Note that I am using a SMART RAMPS board with a MEGA2560.  Both have EEPROM storage capability.  I'd expect the I2C eeprom is being used but now I'm wondering.

'''
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E398.00
echo:; Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Mesh Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 S3 I0 J0 Z-0.30000
echo:  G29 S3 I1 J0 Z-0.27500
echo:  G29 S3 I2 J0 Z-0.55000
echo:  G29 S3 I0 J1 Z-0.20000
echo:  G29 S3 I1 J1 Z-0.22500
echo:  G29 S3 I2 J1 Z-0.57500
echo:  G29 S3 I0 J2 Z0.07500
echo:  G29 S3 I1 J2 Z-0.20000
echo:  G29 S3 I2 J2 Z-0.85000
echo:  G29 S4 Z0.00000
echo:; Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:; PID settings:
echo:  M301 P22.52 I1.91 D66.42
echo:  M304 P52.54 I9.72 D189.38
echo:; LCD Contrast:
echo:  M250 C190
echo:; Z-Probe Offset (mm):
echo:  M851 X-44.00 Y2.00 Z-0.97
echo:; Stepper driver current:
echo:  M906 X650 Y650 Z800
echo:  M906 T0 E550
echo:; StallGuard threshold:
echo:  M914 X70 Y70 Z50
echo:; Driver stepping mode:
echo:  M569 S1 X Y
echo:  M569 S1 T0 E
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U100.00
'''

kpishere, 2021-03-20:
I'd just ran it a fourth time, where I'd had to babystep +2.203 mm last time, This time, after a cold start of perinter, I had to offset +2.380 mm.

Saving to the eeprom without any saving commands?

ellensp, 2021-03-20:
or something like Z max feed rate to high so you get variable Z moves. Ie homes. move up an  amount (eg say 5 mm) but it skips so it does move 5mm up at all. 
Add a M203 Z1  to your start.gcode  (1mm/s  painfully slow, but is only a test)

kpishere, 2021-03-20:
Yeah, I'm wondering if it is something physical.  I see nothing standing out in the code.  I do have springy couplers but they are stretched and then tied down.  I also have it using senseless homing to the Z-MAX position.  

I'd already noted that with this configuration, Z-Height in firmware is very important as any discrepancy from reality is corrected for in the first 10mm (by default) of height.  Actually, now that I think about it, setting the correction height to the Z-MAX height would make that problem negligible but .. digress.

The z-threads are 4-start screws.  I did note that sometimes Z-motion is very quiet but sometimes it isn't quiet.  That could be a loose wire couldn't it.  Ok, I'll focus on that and do some more tests when I'm sure that that is not it.  

I'd just added a 100uF surface mount capacitor on the SMART RAMPS board between the pins of the reset button .. so now, I can connect serial port without triggering restarts.  This will help with debugging.

kpishere, 2021-03-22:
The ticket can now be closed.  What was causing it was a) given I'm doing Z-MAX homing, the defined print height of 255.7mm actually was 253.7mm after I'd replaced the bed springs.  With updating the Z-MAX height to 253.7mm, there is no more crashing.

However, in testing, I'd found some other bugs in developer mode code and in manual probe case.  I'll submit a PR for those and describe more there.

closed 2021-03-22 by kpishere

kpishere, 2021-03-29:
@ellensp Ha!  Discovered the issue with my printer today and what generated this ticket.  It is kind of funny.  I did the no-no, print a head mount in PLA.  Then start printing ABS in a closed chamber.  Learn by doing right.  Of course I knew this was a no no. 

[image]