MarlinFirmware/Marlin #21352: [BUG] Printer tries to restart printing after parking the head when I use Stop Print

Thinkersbluff, 2021-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If I am printing from Cura (4.8) via Octoprint & USB, and if I select Stop Print on the printer before I Abort the print at Cura, then the Printer parks the head, then starts extruding thickly and sluggishly moves back toward the print, for 5-10 seconds.

I have two printers which exhibit this behaviour: 1) a Creality Ender 3 running Marlin 2.0.7.2 and 2) a CR-6 SE running the Community Firmware 6, pre-release 5, which incorporates the latest bugfix2.0.x. 
  
The CR-6 SE has also shown this behaviour when printing from an SD card, so I don't think this is uniquely an Octoprint/USB connection issue.  

### Bug Timeline

I have had this problem for a few months, thinking it was my Octoprint set up.  When it happened from an SD card print, I decided it may actually be a Marlin issue.

### Expected behavior

I expect Stop Print to park the head and stop all print moves. 

### Actual behavior

Instead, the printer seems to continue to process a few more print instructions after parking the head.

### Steps to Reproduce

1. Start a print from Cura 4.8, via Octoprint & USB cable to the printer.
2. Wait for the printer to put-down a layer or two.
3. Select Stop Print, Stop. On the printer menu.
4. The printer will park the head but will then start pushing out filament too quickly and will slowly move back toward the print, before "exhausting" itself.

### Version of Marlin Firmware

On the CR-6 SE, I am using CF6Pre5, which last merged with Bugfix2.0.x on 10 March 2021. On the Ender 3, I am running Marlin 2.0.7.2.

### Printer model

Both Creality Ender 3 (with a BTT SKR Turbo board) and Creality CR6-SE (with a Creality 4.5.2 motheboard)

### Electronics

Creality Ender 3 has a BTT SKR Turbo board. Creality CR6-SE has a Creality 4.5.2 motheboard

### Add-ons

Raspberry Pi3b+ running Octoprint.

### Your Slicer

Cura

### Host Software

OctoPrint

These are my config files:
[CR-6_Configs.zip](https://github.com/MarlinFirmware/Marlin/files/6138785/CR-6_Configs.zip)
[Ender3_Config With BLT.zip](https://github.com/MarlinFirmware/Marlin/files/6138787/Ender3_Config.With.BLT.zip)

thisiskeithb, 2021-03-15:
You have `HOST_ACTION_COMMANDS` & `HOST_PROMPT_SUPPORT` enabled in one config & not the other.

Enable those settings for both configs & reflash. Your printer will pause as it should when printing from OctoPrint.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-03-15 by thisiskeithb

Thinkersbluff, 2021-03-15:
> You have `HOST_ACTION_COMMANDS` & `HOST_PROMPT_SUPPORT` enabled in one config & not the other.
> 
> Enable those settings for both configs & reflash. Your printer will pause/stop as it should when printing from OctoPrint.
> 
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:
> 
> * [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
> * [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
> * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group
> * [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
> * [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
> * [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server
> 
> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Hi, KeithB.
Thank you for the quick reply.  I did start by asking this question [on the Marlin Discord](https://discord.com/channels/461605380783472640/491105274464043026/819983009448984586). 
Maybe I needed to wait a little longer for a reponse.

I don't see any HOST_ACTION or HOST_PROMPT lines in Configuration.h, so I assume by "both configs" you mean I should have also enabled those both in the Ender 3 Configuration_adv.h. Thank you, yes I should.

**PLEASE NOTE**, however, that I also have this issue on the CR-6, which does have both of those enabled, and I have seen this happen when printing from the SD card on my CR-6.  That was the only reason I thought maybe this is NOT a HOST-related problem, per se.

I am content to continue the discussion on Discord, if you prefer.