MarlinFirmware/Marlin #21353: [FR] Enable microsteps configuration on Creality 4.2.7

thegwydd, 2021-03-15:
### Is your feature request related to a problem? Please describe.

I need to set E motor microstepping to 8 on a Creality 4.2.7 board but it seems it can't be done.

### Are you looking for hardware support?

Creality Ender 3 with Flex3Drive and Creality 4.2.7.

### Describe the feature you want

I'd like to set microsteps to 8. I don't care about using gcode 350 (yes, of course it would be better), I can live with a solution that requires re-compiling the firmware.

ellensp, 2021-03-15:
This is a hardware limitation of the controller. It is not settable in firmware at all. You need to physically modify your controller to change the micro stepping.

closed 2021-03-15 by ellensp

thegwydd, 2021-03-15:
wow, e nice news for this monday! 
Is there a walkthrough about the procedure or am I on my own? I can't even find an electronic schematic about Creality 4.2.7 so I don't have a clue about the current hw configuration of controller.

ellensp, 2021-03-15:
Creality do not provide any details on their controllers. no circuit diagrams, not even pin diagrams. All we have is form reverse engineering them.  
Your basically on your own. And I suspect up the creak without a paddle.
The pins you need to get to are under the stepper driver heat sinks. Which can't be removed...

thegwydd, 2021-03-15:
So basically it's more convenient to change the board with one more configurable.
Could the the SKR Mini E3 V2.0 do the job?
At noise level I know it's louder than Creality 4.2.7 but I live with this if it's quieter than Creality stock board.
Do you have any advice about that?

thegwydd, 2021-03-15:
Or may be there are different solutions.
The main problem is that after calibration of steps/mm for the extruder I have 4102 steps/mm to get precise extrusions.
I tried many settings for acceleration, jerk, junction deviation, speed, vref and all settings about the extruder but the extruder motor still squeaks when doing retractions.
I'm using Cura and I tried to reduce acceleration to minimum and the only settings that seems to work without making E motor to skip steps are settings that are not acceptable for printing as retractions are really slow.
From I followed all Flex3Drive documentation directives except what related to microstepping and I was guessing that this could be the main reason.
From my understanding the problem seems to be the way Creality 4.2.7 wants to drive the stepper motor controller that, at the end, uses a microstep of 256 with low current pulses.