MarlinFirmware/Marlin #21356: External EEPROM works fine in Old Bug Fix.

Talha909, 2021-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I had download the new bug fix and add the external EPPROM and some lines in the end of config.adv file. In old bug fix it works good but in latest bug fix it gave me the error message and and did not compile. 

#define MNV_SKRV1_4_512K_END 0xFFFF //64Kb
#define I2C_EEPROM
#define E2END MNV_SKRV1_4_512K_END

### Bug Timeline

in new bug fix. 

### Expected behavior

I expected to work 

### Actual behavior

It does not able to compile. 

### Steps to Reproduce

Download the new bug fix and add external eeprom. And make the modification in the end of config.adv file. But it did not complie. 

### Version of Marlin Firmware

Bugfix 15-3-2021

### Printer model

Ender

### Electronics

SKR 1.4 Trubo with TMC 2130

### Add-ons

Bltouch. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6141587/Marlin.zip)
I download the new bug fix 15-3-2021 and compile it but it did not compile. And gave me the error message. I also write the lines regarding the eeprom. It woks fine in old bug fix. 

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.6
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bltouch.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\cooler.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\powerloss.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\tmc_util.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\tramming.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\z_stepper_align.cpp.o
Marlin\src\HAL\shared\eeprom_if_i2c.cpp: In function 'void _beginTransmission(uint16_t)':
Marlin\src\HAL\shared\eeprom_if_i2c.cpp:55:7: error: 'MARLIN_EEPROM_SIZE' was not declared in this scope
   55 |   if (MARLIN_EEPROM_SIZE > 0x4000) {  // Use two-byte addressing for EEPROMs >16kb
      |       ^~~~~~~~~~~~~~~~~~
Marlin\src\HAL\shared\eeprom_if_i2c.cpp: In function 'uint8_t eeprom_read_byte(uint8_t*)':
Marlin\src\HAL\shared\eeprom_if_i2c.cpp:92:1: warning: no return statement in function returning non-void [-Wreturn-type]
   92 | }
      | ^
*** [.pio\build\LPC1769\src\src\HAL\shared\eeprom_if_i2c.cpp.o] Error 1

thisiskeithb, 2021-03-15:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Talha909, 2021-03-16:
> Bummer this was closed as I have the same issue with SKR Pro 1.1 and 2208s with external i2c eeprom... Seems related to recent changes in eeprom code but I am no developer. :(
> 
> -William

If you find any solution. Please let me know. And if i fond then i will inform you

ellensp, 2021-03-16:
@Talha909 looks like lpc176x code does not use E2END
add this to the end of your Configuration_adv.h 
```
#ifndef MARLIN_EEPROM_SIZE
  #define MARLIN_EEPROM_SIZE size_t(E2END + 1)
#endif
```

ellensp, 2021-03-16:
I think this needs looking into further
 AVR, STM32 and teh TEENSY family HAL's all use E2END, so why doesn't LPC176x HAL..
needs some constancy...

thisiskeithb, 2021-03-16:
> looks like lpc176x code does not use E2END
add this to the end of your Configuration_adv.h

That was done away with some time ago. 

I just helped someone with a similar SKR 1.4 + external EEPROM issue and they were using old instructions which is why I closed this issue.

Old/outdated instructions (where the `MNV_*` define from the original report comes from): https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-v1-4-i2c-eeprom/

Updated instructions: https://www.instructables.com/BigTreeTech-SKR-14-14T-Adding-a-EEPROM/

ellensp, 2021-03-16:
then code needs updated with warning that E2END is obsolete, and the 9 references in bugfix to E2END need fixed...
Either way something needs done. not just closing issue.

chrisjc3, 2021-03-17:
this should be reopened. 

i have an ender 3 pro with skr v2 and bltouch. i got tired of keeping up with firmware so i decided to subscribe to https://marlin.crc.id.au/ 2 days ago and their firmware builds (both builder and nightly) for my exact make and board contain this bug.

marlin firmware compiled from 14feb works as expected. latest doesn't. it's not an uncommon build. this isn't a 'user support issue', you have it wrong on your own official website that you charge money for.

ellensp, 2021-03-17:
@chrisjc3 this is a compile time error. It stops you form building the firmware.  How can you be be getting compile time errors when your not even compiling the code, Your getting precompiled firmwares.,,,

Talha909, 2021-03-17:
[image]
I try to add  these lines 
#ifndef MARLIN_EEPROM_SIZE
  #define MARLIN_EEPROM_SIZE size_t(E2END + 1)
#endif
but still got the compile error.

ellensp, 2021-03-17:
that is a completely different error.
The options you have selected will not fit the  ram of this controller

chrisjc3, 2021-03-17:
@ellensp sorry. it was late and i didn't catch that. 

in february when i did compile it myself all i enabled was EEPROM_SETTINGS and EEPROM_AUTO_INIT and it worked perfect on the same make/board. i didn't need to dig into any of this. i assume this is related but i'm admittedly grasping at straws.

CRCinAU, 2021-03-17:
Just to add info, I've had multiple reports of this issue...

Currently, I'm suspecting this commit:
https://github.com/MarlinFirmware/Marlin/commit/cc3e878f90b38bcd9116c1b4e8ff31a4209299e9

Builds before this commit correctly save to the EEPROM, builds after this commit do not.

CRCinAU, 2021-03-18:
This can probably be closed as a duplicate of: https://github.com/MarlinFirmware/Marlin/issues/21370

I believe the resolution to https://github.com/MarlinFirmware/Marlin/issues/21370 will fix this issue as well.

thinkyhead, 2021-03-18:
Duplicate of #21370

ellensp, 2021-03-18:
still no..  that addresses chrisjc3 unrelated issue, but is not the reason this issue was opened.

CRCinAU, 2021-03-18:
My bad - I got lost in the comments :)