MarlinFirmware/Marlin #21357: PROBE OFFSET {X, Y, Z} inverted if INVERT_X/Y_DIR is true

mrgiov, 2021-03-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When properties INVERT_X_DIR or INVERT_Y_DIR are enabled, the property NOZZLE_TO_PROBE_OFFSET {X, Y, Z} assume inverted values instead of reported in examples written in code lines

### Version of Marlin Firmware

2.0.7.2

### Printer model

Creality CR10S

### Electronics

motherboard BTT SKR MINI E3 V2.0

ellensp, 2021-03-16:
Please attach your configuration files.

mrgiov, 2021-03-16:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6149017/Configuration.h.txt)

hamster65, 2021-03-16:
second config file is missing.

ellensp, 2021-03-17:
Config is not bugfix, which is what 'we' would be testing on....

ellensp, 2021-03-17:
[bugfix-Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/6153657/bugfix-Configuration.h.zip)
Converted to bugfix.

Unrelated to issue. You have two instances of PREHEAT_1 and PREHEAT_2, second occurrence overwrites the first and causes many redefinition warnings.

perhaps you really wanted this?
```
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 185
#define PREHEAT_1_TEMP_BED     45
#define PREHEAT_1_TEMP_CHAMBER 35
#define PREHEAT_1_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_2_LABEL       "ABS"
#define PREHEAT_2_TEMP_HOTEND 240
#define PREHEAT_2_TEMP_BED    110
#define PREHEAT_2_TEMP_CHAMBER 35
#define PREHEAT_2_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_3_LABEL       "petG"
#define PREHEAT_3_TEMP_HOTEND 220
#define PREHEAT_3_TEMP_BED     45
#define PREHEAT_2_TEMP_CHAMBER 35
#define PREHEAT_3_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_4_LABEL       "Nylon"
#define PREHEAT_4_TEMP_HOTEND 230
#define PREHEAT_4_TEMP_BED    45
#define PREHEAT_4_TEMP_CHAMBER 35
#define PREHEAT_4_FAN_SPEED   255 // Value from 0 to 255

 
```

ellensp, 2021-03-18:
still waiting for Configuration_adv.h for bugfix

mrgiov, 2021-03-18:
Don’t worry. Can you close this tiket because it was my mistake.. The Firmware work perfectly Inviato da Posta per Windows 10 Da: ellenspInviato: giovedì 18 marzo 2021 10:07A: MarlinFirmware/MarlinCc: mrgiov; AuthorOggetto: Re: [MarlinFirmware/Marlin] PROBE OFFSET {X, Y, Z} inverted if INVERT_X/Y_DIR is true (#21357) still waiting for Configuration_adv.h for bugfix

closed 2021-05-28 by github-actions[bot]