MarlinFirmware/Marlin #21360: Sensorless Homing Sequence Core XY

fiveorders, 2021-03-16:
### Is your feature request related to a problem? Please describe.

I built a custom Core XY machine and using TMC 2226 drivers with the sensorless homing feature. when I use the default homing speeds such as 50*60, 50*60 axises bump the end and bounce back slightly causing a misalignment which can go up to 4mm on Y-axis. 

### Are you looking for hardware support?

Nope,

### Describe the feature you want

I overcame this issue by reducing speeds to 10*60 during homing but it is annoying slow on 300x300 printer. There are HOMING_BUMP_MM and HOMING_BUMP_DIVISOR features which are lovely, however, the sequence is of this feature not helping.

First, I set the homing speeds to 40*60 which is good enough and set the divisors to 4 so the speed reduces to 10*60 after first bump. But the issue is the sequence of these movements. X homes fast first and step back 5mm and homes slowly and accurately. However, when Y does its sequence, X is jumping a couple of mm at the end. (Neither Bumping sensitivity nor the drive currents can solve the issue)

so the sequence should be like this; X should fast home first, then Y should fast home, these will create draft homing with slight jumps and then divisor should take place and X should home slowly and Y after that.

I hope I was able to tell you the issue. hope it can be fixed. Many thanks in advance

Portzal, 2021-03-18:
It is advised not to use the bump feature with sensor less homing. Instead I use #define HOMING_BACKOFF_POST_MM { 5, 5, 0 }.

fiveorders, 2021-03-18:
As mentioned on my post, that feature does not help due to the sequence of homing. Post backoff should be in order with other axis. Because the last homing axis makes the other axis jump steps. If Y homes last, the first homing action before post backoff makes the X jump several mm right. So post backoff does its job on Y axis but X needs another backoff with slow speed as well.

So again as mentioned, correct order should be (speeds are example)

Homing sequence;
G28 X0 F3600
G28 Y0 F3600
G28 X0 F400 (with 5mm post backoff)
G28 Y0 F400 (with 5mm post backoff)

This way Y and X axis will flush the axis ends without jumping.

Portzal, 2021-03-18:
Are you using the homing current option for X and Y?

fiveorders, 2021-03-18:
These are my current settings right now.

// Homing speeds (mm/min)
#define HOMING_FEEDRATE_MM_M { (15*60), (10*60), (8*60) }

#define SENSORLESS_BACKOFF_MM  { 5, 5 }     // (mm) Backoff from endstops before sensorless homing // ZLC Sensorless

#define HOMING_BUMP_MM      { 0, 0, 0 }       // (mm) Backoff from endstops after first bump // ZLC
#define HOMING_BUMP_DIVISOR { 4, 4, 0 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

//#define HOMING_BACKOFF_POST_MM { 2, 2, 2 }  // (mm) Backoff from endstops after homing

//#define QUICK_HOME                          // If G28 contains XY do a diagonal move first
//#define HOME_Y_BEFORE_X                     // If G28 contains XY home Y before X
//#define HOME_Z_FIRST                        // Home Z first. Requires a Z-MIN endstop (not a probe).
//#define CODEPENDENT_XY_HOMING               // If X/Y can't home without homing Y/X first

Portzal, 2021-03-18:
Sensor less homing settings I am running SKR V1.4 Turbo with TMC2226’s
I am using the following settings that are trouble free and have a reasonably wide tuning band.
Configuration.h
#define HOMING_FEEDRATE_XY (20*60)
Configuration_adv.h
#define HOMING_BUMP_MM { 0, 0, 0 }
#define HOMING_BACKOFF_POST_MM { 5, 5, 0 }
#define X_CURRENT_HOME (1/2 to 3/4 of full current)
#define Y_CURRENT_HOME (1/2 to 3/4 of full current)
#define SENSORLESS_HOMING
#define X_STALL_SENSITIVITY (tuned to machine)
#define Y_STALL_SENSITIVITY (tuned to machine)
I found #define QUICK_HOME caused issues with sensor less homing, if Y homed before X, Y would hold on and not back off until X homed. Leading to noisy and inconsistent homing.
I keep sensitivity settings for X and Y the same. 
I found too low a homing current can cause false triggers.
These settings will also work with a higher homing speed.

fiveorders, 2021-03-18:
I will try altering the X Current Homing value now thank you for the idea.

BTW, #define HOMING_BACKOFF_POST_MM { 5, 5, 0 } Why do you need this 5mm?

> On 18 Mar 2021, at 08:55, Portzal ***@***.***> wrote:
> 
> #define HOMING_BACKOFF_POST_MM { 5, 5, 0 }

fiveorders, 2021-03-18:
I tried reducing the Currents when homing to the bottom limit but it doesn’t help. Speed is the issue. So as I mentioned in my previous comments it should be in such sequence.

Otherwise we need to use 10*60 for Y and 15*60 for X :(

Portzal, 2021-03-18:
I use 5mm back off to give some margin for a re home. I found it gives more consistent results than if you try to home already on the end stops.

organizedkaos, 2021-03-25:
> I tried reducing the Currents when homing to the bottom limit but it doesn’t help. Speed is the issue. So as I mentioned in my previous comments it should be in such sequence. Otherwise we need to use 10*60 for Y and 15*60 for X :(

You don't reduce the homing current when doing sensorless homing.  You want your x/y currents for normal printing to be at the lowest possible without losing steps.  Then you set your homing current slightly above that.

Everything ties together when it comes to sensorless homing.  I can use 75*60 homing feedrate on my corexy x/y axis but it doesn't really do much since my max feedrates are capped at 1500.

When you get your currents right you have to set your stall sensitivity based on the total speeds you are using.

Portzal, 2021-03-25:
CURRENT_HOME reduces current for homing moves only. It is a very handy setting and can greatly improve Sensor less homing reliability. [image]

fiveorders, 2021-03-25:
Yes but that doesn’t help because it is causing false triggers. The only solution will be slow speed instead of low current.

Portzal, 2021-03-25:
Too low a homing current can cause false triggers, if you have it set lower than normal current but high enough to avoid false triggers it gives you a softer homing bump.

organizedkaos, 2021-03-25:
Have you thought about providing your config files for us to review?

My currents for my x/y are around 580 and my home current is at 800.  0 issues with sensorless homing and max feedrates at 1500.  Obviously I'm not going for speed anyways.

fiveorders, 2021-03-25:
My X Y currents are 1000 right now but I have started with 600. Homing currents are same now but I tried several different combinations none of them helped.

If the currents are low or sensivity is high I am getting false triggers, and one axis makes the other axis jump several mm if the speed is over 15*60

Now I reduced the speeds to X/15*60 and Y/10*60 and it works without making the other axis jump. So what I need is to change the sequence of X Y calibration. First X/50*60 and Y/50*60 then 5mm retraction and X/10*60 and then Y/10*60.

organizedkaos, 2021-03-25:
I have 2 coreXY machines and a custom coreXZ machine all working on sensorless homing for the appropriate axis.  They all use the same homing speeds for the applicable axis without issues.

Either post your configs for us to review or check your mechanical.

fiveorders, 2021-03-25:
#if AXIS_IS_TMC(X)
    #define X_CURRENT       900        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  X_CURRENT        // (mA) RMS current for sensorless homing 
    #define X_MICROSTEPS     16        // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1        // -1..0: Not chained. 1: MCU MOSI connected. 2: Next in chain, ...
    //#define X_INTERPOLATE  true      // Enable to override 'INTERPOLATE' for the X axis
  #endif

#if AXIS_IS_TMC(Y)
    #define Y_CURRENT       900
    #define Y_CURRENT_HOME  Y_CURRENT       // (mA) RMS current for sensorless homing 
    #define Y_MICROSTEPS     16
    #define Y_RSENSE          0.11
    #define Y_CHAIN_POS      -1
    //#define Y_INTERPOLATE  true
  #endif

#define SENSORLESS_BACKOFF_MM  { 5, 5 }     // (mm) Backoff from endstops before sensorless homing 

#define HOMING_BUMP_MM      { 0, 0, 0 }       // (mm) Backoff from endstops after first bump 
#define HOMING_BUMP_DIVISOR { 4, 4, 0 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

// Homing speeds (mm/min)
#define HOMING_FEEDRATE_MM_M { (15*60), (10*60), (8*60) } 

These are my current configuration values. Speed divisors are ineffective as bump values are 0.

Again, my problem is if I home with speeds above 15*60 X bump makes Y jump 1-4 mm or vice versa, I don’t know what to check on hardware.

Portzal, 2021-03-25:
@organizedkaos I am curious as to why you would increase current for homing. My understanding and practical experience is reducing current for homing moves reduces torque being delivered by the stepper and results in a softer bump and wider sensitivity tuning band.

organizedkaos, 2021-03-25:
> @organizedkaos I am curious as to why you would increase current for homing. My understanding and practical experience is reducing current for homing moves reduces torque being delivered by the stepper and results in a softer bump and wider sensitivity tuning band.

You know I thought I had notes or a link to information I thought I read regarding higher homing currents for sensorless homing.  I can't seem to find it.  All I know is that for my tmc2209s I got early/false triggers at lower homing currents.

Portzal, 2021-03-26:
There is a point where too low causes issues such as false triggers. But even a 200-300mA drop from running current can make a difference in improving homing reliability. I have TMC2209's and TMC2226's.

fiveorders, 2021-03-26:
Lower currents for homing improves the homing reliablity however does not fix in my case. Speed is my problem. 

Does anyone know which file contains the homing sequence codes? I want to try changing the sequence as below:

G28 X F3600
G28 Y F3600
G28 X F600
G28 Y F600

fiveorders, 2021-07-20:
> I have 2 coreXY machines and a custom coreXZ machine all working on sensorless homing for the appropriate axis. They all use the same homing speeds for the applicable axis without issues.
> 
> Either post your configs for us to review or check your mechanical.

Can you post your configuration here?

I have tried everything but whichever axis homes last, makes the other axis jump a random distance between 0.5m-3mm

thisiskeithb, 2021-07-25:
@fiveorders: You opened this as Feature Request, but it sounds more like you’re looking for help with getting sensorless homing setup.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-07-25 by thisiskeithb

fiveorders, 2021-07-25:
Actually it is a requirement. Whatever I did I couldn’t achieve a reliable Sensorless Homing on CORE XY system. Finally moved back to end stops. There are several other posts about this issue such as the other motor is not monitored during homing etc. I am not sure if it is technically possible to listen and manage two motors working together during homing a single axis. The only way to do a faster homing is what I mentioned above. Changing the homing bump back sequence. 

As far as I see sensorless homing is not prefered on core XY, or it is still an experimental feature. This is what I have got from forums and some experienced people. 

Thank you anyway, sorry for asking help in the same post. Cheers