MarlinFirmware/Marlin #21367: Problem trying to calibrate a BIG delta and G33

AlvaroRey, 2021-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

A few years ago I assemble a big delta printer 1.5 meters in diameter and 1 meter hight print volume, at first I try to autocalibrate using UBL and a FSR probe, the first run of the G33 was ok, no problem at all, but when the machine goes homing and start the second round, I have to turn it off,  the height measure is correct, but when the machine go for measure the tower, it start to go up and I got to turn the machine off because the effector crash in first the tower. From time to time I try different version of marlin just in case, but it happen consistenly trough all the version, the last one testes 2.0.6 last week, same problem
I put into the marlin 1.1.6 all the data, delta diagonal rod and all that stuff, I was able to calibrate the delta parameters, so the machine

### Bug Timeline

This is an old issue, 2 years ago when I assemble the machine, it was imposible to autocalibrate with a FSR probe

### Steps to Reproduce

I think its going to be dificult to reproduce unless you have such a delta printer 3 meter hight, 2 meter in diameter.

But just need to use G33 P2 and the problem appear, I can't go higher thar P2

### Version of Marlin Firmware

2.0.6 right now

### Electronics

Ramps 1.4 + arduino mega

### Your Slicer

Slic3r

### Host Software

Repetier Host

This are my config files [https://www.dropbox.com/s/hq8wxi5f5plji0y/Configuration.h?dl=0 https://www.dropbox.com/s/hqrgf5nc4nkx8kg/Configuration_adv.h?dl=0]()

thisiskeithb, 2021-03-17:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

LVD-AC, 2021-10-01:
This could be related to #22853

thisiskeithb, 2021-10-10:
There was no reply from OP, but we now have multiple issues for the same bug. Please move discussions to https://github.com/MarlinFirmware/Marlin/issues/22886