MarlinFirmware/Marlin #21368: skr pro 1.2 bed levelling tests only 1 point and stops no test 9

subzero2379, 2021-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

skr pro 1.2 bed levelling tests only 1 point and stops no test 9 

### Actual behavior

tests only point 1

### Steps to Reproduce

install fw on skr pro with bilinear bed levelling on

### Version of Marlin Firmware

2.0.7.2

### Printer model

tatara frame based oin anet a8

### Electronics

skr pro 1.2

### Your Slicer

Simplify3D

skr pro 1.2 bed levelling tests only 1 point and stops no test 9 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6158866/Marlin.zip)

thinkyhead, 2021-03-17:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

GhostlyCrowd, 2021-03-18:
> ### Did you test the latest `bugfix-2.0.x` code?
> Yes, and the problem still exists.
>
> ### Bug Description
> skr pro 1.2 bed levelling tests only 1 point and stops no test 9
>
> ### Bug Timeline
> _No [...]

These configs are from release 2.0.7.2 please upload your bugvid-2.0.x configs I have the same platform and have no issues os I'm curious for your configs.

subzero2379, 2021-03-19:
download latest bugfix and its ok

closed 2021-03-28 by thisiskeithb