MarlinFirmware/Marlin #21370: [BUG] (EEPROM version mismatch) SKR Mini E3 2.0

iZz0s, 2021-03-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

EEPROM isn't working after compiling Firmware for SKR Mini E3 2.0 (bugfix-2.0.x branch). I got everytime a "EEPROM version mismatch" after tryling to load.

Using "STM32F103RC_btt" as default_envs (tried 512k too) and Motherboard defined as "BOARD_BTT_SKR_MINI_E3_V2_0".

#define EEPROM_SETTINGS is enabled.

It was working with the 2.0.7.2 Firmware from BTT github repo:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

EEPROM should be a AT24C32 after looking for details.

### Expected behavior

Store/Load EEPROM should working on oboard AT24C32 EEPROM.

### Actual behavior

Store/Load EEPROM isn't working.

### Steps to Reproduce

1. FLASH bugfix FW to SKR Mini E3 v2.0
2. Try to Store/Load EEPROM.

### Version of Marlin Firmware

bugfix-2.0.x newest

### Printer model

Sovol SV01

### Electronics

SKR Mini E3 v2.0

### Add-ons

BLTouch

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

CRCinAU, 2021-03-17:
To test if this is an issue or not, reset the EEPROM contents via M502, then save with M500.

If its not an actual issue, and just your first time upgrading it to bugfix, this may well be expected behaviour when the EEPROM settings structure changes.

iZz0s, 2021-03-18:
I've tried to reset via M502 and saved then with M500. After trying to load (or restart printer) i got the same error again.

i was testing with #define EEPROM_AUTO_INIT before, too.

thinkyhead, 2021-03-18:
For debugging, add to your configuration:
```cpp
#define DEBUG_EEPROM_READWRITE
```
And then after a new flash of the firmware, you will get some extra detailed messages when the EEPROM is loading or saving. If you see an error message then it means we probably have a bug or typo in the firmware.

iZz0s, 2021-03-18:
Send: 2:44:48.478: N17 M502
Recv: 2:44:49.361: ok
Send: 2:44:51.095: N18 M501
Recv: 2:44:51.982: ok
Send: 2:44:53.461: N19 M500
Recv: 2:44:53.468: echo:Error writing to EEPROM!
Recv: 2:44:53.469: Error:Field esteppers mismatch.
Recv: 2:44:53.861: echo:Error writing to EEPROM! (63)
Recv: 2:44:53.862: ok

thisiskeithb, 2021-03-18:
> For debugging, add to your configuration:
> 
> ```c++
> #define DEBUG_EEPROM_READWRITE
> ```
> 
> And then after a new flash of the firmware, you will get some extra detailed messages when the EEPROM is loading or saving. If you see any extra error messages then it means we probably have a bug or typo in the firmware.

@thinkyhead I can confirm that EEPROM seems to be broken on several builds here in my lab space after #21337. Here's some `M502`/`M500` output from an SKR E3 Turbo with an onboard AT24C32 for example:

<details><summary>Full error with latest bugfix-2.0.x (deaefbf):</summary>

```prolog
Send: M502
Recv: ok
[...]
Send: M500
Recv: echo:Error writing to EEPROM!
Recv: Error:Field esteppers mismatch.
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: echo:Error writing to EEPROM!
Recv: ok
```
</details>

<details><summary>After dropping "Fix small wired EEPROM (#21337)" (cc3e878)</summary>

```prolog
Send: M502
Recv: ok
[...]
Send: M500
Recv: //action:notification Settings Stored
Recv: ok
```
</details>

thinkyhead, 2021-03-18:
So, evidently the changes from 21337 apply to some models of EEPROM but not to others. The changes reminded me of the BL24CXX that I had added separately. Maybe what was needed was to make a copy of that EEPROM and modify it for the new model being introduced.

Meanwhile, we can revert 21337 and then interrogate the original author to figure out the best course of action.

iZz0s, 2021-03-18:
i can confirm that it works with an older Marlin/src/HAL/shared/eeprom_if_i2c.cpp.

I've used the content from here:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/Marlin-2.0.7.2-SKR-mini-E3-V1.2/Marlin/src/HAL/shared/eeprom_if_i2c.cpp

now my output is like this:
Recv: 3:34:35.068: ok
Send: 3:34:42.493: N18 M502
Recv: 3:34:43.378: ok
Send: 3:34:45.140: N19 M500
Recv: 3:34:45.511: ok

so the problem seems to be the change from 21337.

thisiskeithb, 2021-03-18:
Closing since the breaking commit was reverted: https://github.com/MarlinFirmware/Marlin/commit/560448afeddfdea568d74c5be4e48c8b7f5886c7

closed 2021-03-18 by thisiskeithb