MarlinFirmware/Marlin #21371: Invert thermistor working (if temperature higher activate thermistor)

kekko7072, 2021-03-17:
### Describe the feature you want

Hey is there a way to simply invert the logic of thermistor working in Marlin software for 3D Printing. In fact I'd like to run current to the thermistor port (that I am using for fan) if the temperature is higher than configured, and stop current if the temperature is lower, because due to the heat it will rise on it's own. Is there a simple way to do it? I need 5V 1.5 A for this fan.

No specific board is selected for this project, I can use all type of board.

ellensp, 2021-03-18:
look at auto_fans
turned on over certain temp, off when drops under that temp

Rincewind08, 2021-03-28:
i alreqady do this running pumps.   use a heater port to make sure that you have enough amperage in your .pins, if you have a heater port to spare.
`

closed 2021-04-16 by kekko7072