MarlinFirmware/Marlin #21377: [BUG] Typo in lcd/dwin/e3v2/dwin.cpp PSTR("G28O\nG29")

enderextender, 2021-03-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I think this is a bug.  dwin.cpp in lcd/dwin/e3v2

```
#if HAS_ONESTEP_LEVELING

  /* Leveling */
  void HMI_Leveling() {
    Popup_Window_Leveling();
    DWIN_UpdateLCD();
    queue.inject_P(PSTR("G28O\nG29"));
  }

#endif
```

the PSTR says G280 but it should be G28

### Expected behavior

Auto level menu should issue G28 followed by G29, however, the G28 string has a typo preventing homing and thus will crash the printer if user has not homed first.

### Actual behavior

On ender 3 v2 LCD screen, go to Auto Level.  It starts auto level process without homing.  

### Steps to Reproduce

On ender 3 v2 LCD screen, go to Auto Level.  It starts auto level process without homing.  

### Version of Marlin Firmware

2.0.x-bugfix

### Printer model

Ender 3 V2

### Electronics

Creality 4.2.2 board

/* Leveling */
void HMI_Leveling(void) {
  Popup_Window_Leveling();
  DWIN_UpdateLCD();
  queue.inject_P(PSTR("G28O\nG29"));
}

GMagician, 2021-03-18:
G28 O exists, it homes only if home has not been done

ellensp, 2021-03-18:
That is not a bug
The command is G28 O  (capital O)
G28 - Auto Home, [O] Optional. If the position is known then exit without homing.

closed 2021-03-18 by ellensp

enderextender, 2021-03-19:
So G28O with no spaces is acceptable?

ellensp, 2021-03-19:
yip, spaces are purely for human readability, they are not required. Some new serial protocols, for eg meatpack,  remove spaces where it can, to stop wasting time sending them to the controller.