MarlinFirmware/Marlin #21378: [BUG] (Manual bed leveling ignores all firmware speed settings)

QuestionDevelopment, 2021-03-18:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When running manual bed leveling (moves nozzle to four corners of bed) the speed settings in firmware are ignored.  I added another z axis to my print and it can not handle the extremely fast speeds and is causing skipped steps.  For for my situation I am referring to the settings DEFAULT_MAX_FEEDRATE and/or HOMING_FEEDRATE_Z.

### Bug Timeline

New

### Expected behavior

I expect the speed to be controlled by either DEFAULT_MAX_FEEDRATE and/or HOMING_FEEDRATE_Z.

### Actual behavior

Speed is extremely fast.

### Steps to Reproduce

Use manual bed leveling.  The speed is always extremely fast.

### Version of Marlin Firmware

2.0.7.2

### Printer model

Creality Ender 5

### Electronics

Bigtreetech SKR Mini E3 v2

### Add-ons

None

### Your Slicer

Cura

### Host Software

OctoPrint

Configuration Files : 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6167725/Configuration.zip)

thisiskeithb, 2021-03-18:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

QuestionDevelopment, 2021-03-19:
I don't know the process of "upgrading" the firmware.  Can I utilize the same configuration.h, configuration_adv.h and platoformio.ini files?  I'm using BigTreeTech release currently which I have no idea what other changes/libraries included within it...how am I suppose to handle this?

ellensp, 2021-03-19:
"BigTreeTech release" is not current marlin, it is their badly maintained fork of marlin. 
We cannot maintain their fork of marlin.   

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-03-19 by ellensp