MarlinFirmware/Marlin #21379: Help develop gcode for BTT Smart filament sensor

Ruissilva, 2021-03-19:
### Is your feature request related to a problem? Please describe.

Only M600 for pause feature

### Are you looking for hardware support?

No

### Describe the feature you want

Very simple add-on

Hello, I want help creating a Gcode, M601.

Something like:

M601{
Go to pause position
LCD: "Do filament change? Yes or No"
If Yes: Do M600
If No: Resume Print.
}

This will help a lot of people using the new smart filament sensor, because this way we dont have to change the filament if we have a clogged nozzle or a spool jam. 

Can you please help?

Thank you in advance

GhostlyCrowd, 2021-03-19:
If you want to pause without doing a change you should be using M125 i believe `If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.`

Ruissilva, 2021-03-19:
Yes and No. Yes I want to be able to park without the filament change, so M125 makes sense. But I also want to be able to change the filament.  So, only M125 will be incomplet. Also M600 is incomplete.
The BTT smart filament sensor only sends 0's and 1's and cannot distinguish when it is only supposed to pause or when the filament has runout and it is supposed to change filament. I Think a good solution is to provide the user the option to choose, so i think the printer should go to park position and then send a message to the user asking if he wants to resume print or change filament.

Ruissilva, 2021-03-19:
I may be complicating things, i still have no user experience with this sensor. I will try use it always doing filament change, using M600, and see how that goes, but the concept of always having to do a filament change, no matter what, is strange for me. 

I also had the ideia to create a M601, because of the octoprint plugin with this sensor. Ideally i whould connect it to the raspberry, but the only options of the plugin in cause of pause are:

M0, M1 - Give a error message.
M25 - Enters in pause and it is not possible to remove pause.
M226 - Nothing happens.
M600 - Switch filament. It is not a pause command.
M601 - Unknown command.

So i thought on creating a M601 customized where i could do choose what i want to do, if just resume and everything is ok or efectively there was a filament runout situation and it has to change the filament. Using the M601, I could use the plugin. If I knew how to code marlin, I whould definitly do this, so I asked your help with this

Ruissilva, 2021-03-24:
Ok, I am testing the smart filament sensor with M600 and I am happy. You can close the issue. Thanks for the help.

closed 2021-09-24 by thisiskeithb

mydevpeeps, 2021-09-24:
@thisiskeithb I actually would like to second this original request. Can we re-open it or do you want me to clone it. When printing if if the head is too close to the bed it causes the filament to stop.. and this causes the smart sensor to trigger, extract a perfectly good filament 700mm backwards and then wait for me to say load the same filament again. 

@Ruissilva I'd like the option as well to choose as well and an M601 instead of an M600 is a great way to achieve this.  If someone is using a runout sensor switch and it triggers.. there's no filament there. Use M600. If they are using a smart sensor, we don't actually know if the filament ran out.. we just know it stopped moving. The other thing about M600 is it gives you no other option but to load/unload. So, if it's too close to the bed, clogged (and you need to up the temp 5C).. you can't do any of that. You can only load/unload filament and let it continue on where it left off, failing for the same reason.. and it repeats in an endless loop of uselessness. 

Based on the original OP I suggest M601 with the following options (configurable by CONFIGURATION_CONSTANTS):
M601 will automatically trigger pause print (and park the head if that's configured).
A menu appears with the following options:
- Change filament (triggers M600)
- Change Babysteps (if enabled)
- Change Temperature
- Stop (kills job)
- Continue (resumes print)

This would accommodate almost all scenarios for those using a smart sensor, when the filament is actually not "runout". 

I a very rusty in C but I'd be willing to give it a try with a little help from someone who knows the core better than i do. I Could start by cloning M600 into an M601 and poke around to see how to trigger the different options.. 

Thoughts?

thinkyhead, 2022-04-21:
@mydevpeeps — #23899