MarlinFirmware/Marlin #21382: [BUG] (bug summary)

Demitryk, 2021-03-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The Tevo Tornado _Bootscreen file does not work

### Bug Timeline

Old issue. for as long as i have been pulling those files 

### Expected behavior

Show the default boot screen.

### Actual behavior

Show a garbled boot screen

### Steps to Reproduce

Enable custom bootsceen in configuration.h
add the _bootscreen.h to marlin directory
write firmware to board

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Tevo Tornado

I fixed it, it just needs to be uploaded to the github example directory

[Working Stock Tevo Tornado _Bootscreen.h.zip](https://github.com/MarlinFirmware/Marlin/files/6173315/Working.Stock.Tevo.Tornado._Bootscreen.h.zip)

Vertabreak, 2021-03-19:
if you have a fix why not file a pull request with the change rather then a bug report.

Demitryk, 2021-03-19:
ive never done that before, ill look in to it, and delete this

closed 2021-03-19 by Demitryk