MarlinFirmware/Marlin #21385: [BUG] FYSETC Mini12864 Panel / BigTreeTech Mini12864 V1.0 blank screen

Markus-1984, 2021-03-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Blank screen after bootup

### Expected behavior

Display printer status

### Actual behavior

blank screen

### Steps to Reproduce

powerup

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Custom Cartesian

### Electronics

BigTreeTech GTR V1.0

Description
For the FYSTEC Mini12864 and BTT Mini12864 V1.0 the LCD_CONTRAST has to be set to 255. Otherwise the display will stay blank after the bootup logo because setContrast is not working for Mini Panel.

Compare to https://wiki.fysetc.com/Mini12864_Panel/

Requirements
FYSETC Mini12864 Panel or BigTreeTech Mini12864 V1.0

Benefits
Fix blank screen

Configurations
Conditionals_post.h

change from
#elif ENABLED(FYSETC_MINI_12864)
#define _LCD_CONTRAST_INIT 250

to
#elif ENABLED(FYSETC_MINI_12864)
#define _LCD_CONTRAST_MIN 255
#define _LCD_CONTRAST_INIT 255

thisiskeithb, 2021-03-19:
With `_LCD_CONTRAST_MIN` set to 255, are you able to reduce contrast below 255?

Markus-1984, 2021-03-19:
No, because the setContrast function does not work with display. Refer to the link I've posted.

rhapsodyv, 2021-03-19:
related to #21264 and #20901 ??

Markus-1984, 2021-03-20:
It´s not dependet to the delay. It´s just the contrast of the display. Please refer to https://wiki.fysetc.com/Mini12864_Panel/

thisiskeithb, 2021-11-29:
Keep in mind that `FYSETC_MINI_12864` != `FYSETC_MINI_12864_2_1`, so you're looking at the wrong LCD type. The [BTT Mini 12864 V1](https://github.com/bigtreetech/BGTHREETECH-MINI-12864-V1.0/blob/main/Hardware/Bigtreetech%20Mini12864%20V1.0%20User%20Manual.pdf) manual instructs you to enable `FYSETC_MINI_12864_2_1` and edit some core files, but you no longer have to do that.

Also, with https://github.com/MarlinFirmware/Marlin/pull/23130 merged, you can now simply enable `BTT_MINI_12864_V1` and it will enable the correct settings for you.

_(Remember to initialize EEPROM after flashing or your LCD will likely be unreadable)_

closed 2021-11-29 by thisiskeithb