MarlinFirmware/Marlin #21386: adding Robin Nano V3

Liriel-666, 2021-03-19:
### Are you looking for hardware support?

Please insert Support for the MKS Robin Nano v3.

### Describe the feature you want

Please insert Support for the MKS Robin Nano v3.

Please insert Support for the MKS Robin Nano v3.
its only V1 and V2 in the Files buit V3 is other then these 2

thisiskeithb, 2021-03-19:
Support for this board was added  in https://github.com/MarlinFirmware/Marlin/pull/20430.

closed 2021-03-19 by thisiskeithb

Liriel-666, 2021-03-19:
it was added!
but now it is removed!
the build of today has noo robin nano v3 in it

in Marlin/src/pins/stm32f4/ is no robin nano v3 file
even in boards.h it is removed!

Liriel-666, 2021-03-19:
> 
> 
> Support for this board was added in #20430.
> 
> Use `BOARD_MKS_ROBIN_NANO_V3` as the motherboard define.

it was added!
but now it is removed!
the build of today has noo robin nano v3 in it

in Marlin/src/pins/stm32f4/ is no robin nano v3 file
even in boards.h it is removed!

rhapsodyv, 2021-03-19:
`BOARD_MKS_ROBIN_NANO_V3` is the name of the board. The env in PlatformIO.ini should be one of these: mks_robin_nano_v3, mks_robin_nano_v3_usb_flash_drive, mks_robin_nano_v3_usb_flash_drive_msc.

Liriel-666, 2021-03-19:
all 3 is unknown when put in the Platfoio.ini!
please look in the current Github!
no Robin Nano V3 in Boards.h and no File in any of the Pin folders.
someone removed it!

"> Executing task: C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe run <

Error: Unknown environment names 'mks_robin_nano_v3'. Valid names are 'mks_robin, STM32F103RC_btt, STM32F103RE_btt, melzi_optimized, FYSETC_F6_13, teensy36, LPC1769, mks_robin_stm32, DUE, STM32F103CB_malyan, BIGTREE_SKR_PRO, FYSETC_F6_14, LPC1768, at90usb1286_dfu, LERDGEK, mks_robin_pro, mks_robin_lite3, DUE_debug, chitu_f103, STM32F103RC, STM32F407VE_black, malyan_M300, rambo, LERDGES, MightyBoard1280, STM32F103RC_btt_512K, NUCLEO_F767ZI, jgaurora_a5s_a1, STM32F070RB_malyan, chitu_v5_gpio_init, sanguino644p, FLYF407ZG, teensy31, include_tree, DUE_USB, ARMED, STM32F103RE_btt_USB, SAMD51_grandcentral_m4, mega1280, STM32F103RET6_creality, STM32F103VE_GTM32, STM32F103RC_btt_USB, mks_robin_lite, FLY_MINI, at90usb1286_cdc, STM32F070CB_malyan, LERDGEX, linux_native, mks_robin_nano35, DUE_archim_debug, mega2560, mks_robin_e3, melzi_optiboot, MightyBoard2560, BIGTREE_BTT002, esp32, mks_robin_e3p, sanguino1284p, STM32F103RC_meeb, mega2560ext, STM32F103RC_fysetc, teensy35, STM32F4, melzi, mks_robin_mini, trigorilla_pro, STM32F103RE, DUE_archim, STM32F401VE_STEVAL, FYSETC_S6, rumba32, STM32F7, BIGTREE_GTR_V1_0, teensy41, STM32F103RC_btt_512K_USB, STM32F103VE_longer'
The terminal process "C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe run <

Error: Unknown environment names 'mks_robin_nano_v3_usb_flash_drive'. Valid names are 'mks_robin_stm32, esp32, DUE, mks_robin_e3, mks_robin, mega1280, STM32F103VE_longer, MightyBoard2560, rumba32, DUE_USB, STM32F103CB_malyan, teensy41, mks_robin_lite3, STM32F070CB_malyan, mega2560, mega2560ext, FYSETC_F6_13, linux_native, LERDGEX, NUCLEO_F767ZI, STM32F103RE, DUE_archim, BIGTREE_GTR_V1_0, STM32F103RC_btt_512K_USB, rambo, at90usb1286_dfu, mks_robin_mini, STM32F103VE_GTM32, trigorilla_pro, jgaurora_a5s_a1, STM32F103RC_fysetc, malyan_M300, FLY_MINI, FYSETC_S6, teensy35, STM32F070RB_malyan, STM32F401VE_STEVAL, mks_robin_nano35, STM32F103RC_meeb, teensy31, chitu_v5_gpio_init, sanguino1284p, LERDGEK, include_tree, SAMD51_grandcentral_m4, FYSETC_F6_14, melzi, STM32F407VE_black, LPC1769, STM32F103RE_btt_USB, melzi_optimized, FLYF407ZG, DUE_archim_debug, STM32F103RC_btt_USB, mks_robin_lite, BIGTREE_BTT002, STM32F103RC, at90usb1286_cdc, melzi_optiboot, STM32F103RC_btt_512K, teensy36, mks_robin_pro, sanguino644p, STM32F7, ARMED, STM32F103RET6_creality, MightyBoard1280, LERDGES, STM32F103RC_btt, DUE_debug, chitu_f103, BIGTREE_SKR_PRO, STM32F4, LPC1768, STM32F103RE_btt, mks_robin_e3p' 
The terminal process "C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe run <

Error: Unknown environment names 'mks_robin_nano_v3_usb_flash_drive_msc'. Valid names are 'mks_robin_mini, STM32F103RC_btt_512K, mks_robin_lite, mks_robin_lite3, LERDGES, STM32F103RC, STM32F103VE_longer, mks_robin, FLY_MINI, teensy36, teensy41, trigorilla_pro, mega1280, STM32F103RET6_creality, teensy31, rumba32, sanguino644p, STM32F103RC_btt_512K_USB, jgaurora_a5s_a1, melzi_optimized, DUE_debug, teensy35, STM32F103RC_fysetc, SAMD51_grandcentral_m4, STM32F103RC_btt, linux_native, mega2560, STM32F103RC_btt_USB, malyan_M300, STM32F4, FLYF407ZG, LERDGEX, chitu_f103, STM32F103RE_btt, mks_robin_stm32, at90usb1286_dfu, sanguino1284p, LPC1769, FYSETC_S6, BIGTREE_BTT002, esp32, mks_robin_nano35, STM32F070RB_malyan, STM32F103RC_meeb, STM32F103RE_btt_USB, STM32F103CB_malyan, LERDGEK, rambo, STM32F7, chitu_v5_gpio_init, STM32F407VE_black, STM32F401VE_STEVAL, include_tree, ARMED, DUE_archim, mega2560ext, at90usb1286_cdc, MightyBoard2560, melzi, DUE_archim_debug, STM32F103VE_GTM32, mks_robin_e3, mks_robin_e3p, STM32F070CB_malyan, mks_robin_pro, BIGTREE_GTR_V1_0, DUE_USB, FYSETC_F6_13, NUCLEO_F767ZI, FYSETC_F6_14, STM32F103RE, LPC1768, BIGTREE_SKR_PRO, DUE, MightyBoard1280, melzi_optiboot'
The terminal process "C:\Users\Dark-\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

rhapsodyv, 2021-03-19:
All are there. Use bugfix branch.

Liriel-666, 2021-03-19:
in the bugfix but not in the normal github build

thisiskeithb, 2021-03-19:
> in the bugfix but not in the normal 2.0.x github build.
> then i use that

Support for the V3 was added to `bugfix-2.0.x` after `2.0.7.2` was released. Once the next version of Marlin is ready, support for the V3 board will be included.

Until then, use `bugfix-2.0.x` if you need immediate support for this board.