MarlinFirmware/Marlin #21389: [FR] Control board with SPI with shift registers? instead of step/dir/enable pins for each motor

djamu, 2021-03-20:
### Describe the feature you want

Hi all, I'd like to have your opinion about following feature/concept.

I was wondering if anybody ever tried an all/only SPI build of Marlin, using (high speed) shift registers/transceivers ( either 8 / 16 bit ).
I'm prototyping a control board like that, as it greatly simplifies board layout, daisy chaining SPI shift registers.
Common sense dictates to use buffer IC's, before driver IC's, so why not using an SPI bus?
Modern devices like the IMXRT1060 (teensy 4) etc... all feature high speed SPI bus(ses).
With those speeds, toggling pins on the shift registers can be quicker than those onboard when using DMA.
80Mhz / Mbs can be achieved, theoretically, without the overhead using 8 8 bit registers, (64 pins), this would be 625kHz on all pins. Which is more then fast enough to drive STEP/DIR pins, and even a localized SPI bus.

I've only glanced briefly at the code, but practically, I think the easiest to achieve this is to map the pins to a Union, toggle bits there and transfer its bytes at a fixed frequency using DMA block transfers.

This is actually a similar approach as to PC motherboards that use high speed serial connections instead of parallel.
Aside from the simplified board layout, which is modular in design, one could use cpu's with fewer pins yet fast enough to drive lots of motors / mosfets. I think I'd want to keep the analog in though.....

My coding skills are sufficient, as I can understand most code, but I'd like to discuss the best approach before going there.
Currently I have assigned 8 bits I/O / motor / other.  ( STEP/DIR/ 4 SPI / enable / endstop ), analog inputs are still routed to main CPU ADC's.

good idea ?

ejtagle, 2021-03-24:
This idea was also discussed for the ESP32 implementation of Marlin. The ESP32 has simply not enough pins to control everything, so a SPI bus expander (shift register+latch) was proposed...

djamu, 2021-03-24:
Are you referring to #11873 ? as I can't find any other issue / feature request related. Aforementioned is not what I mean.

But indeed a ESP32 is a good candidate too. Does the ESP32 port compiles for both CPU's? 
If it doesn't I might actually use an ESP32, ( I've worked almost exclusively with it over the last few years ).
I would then be able to write a driver for the (idle) 2nd core, with a minimum of alterations to marlin itself, ( I wouldn't use wifi/bt)
The result would be a kind of wrapper for marlin... quite ugly... not really what the devs would have in mind I quess ;-) 

On the other hand, the IMXRT1060 is much more powerful, has a feature called FlexIO which would be ideal for creating individual SPI busses after the shift registers. A good example of this is [https://github.com/wramsdell/TriantaduoWS2811](url)

But I'm not familiar enough (yet) with marlin code, lots of reading to do.
I'm looking for some guidance in what / where to start, 

So if anyone wants to help get me started I'd appreciate it.