MarlinFirmware/Marlin #21390: Extruder motor stops mid print (not Linear advance related)

willmackey2nd, 2021-03-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm having an issue with the extruder motor stopping near the end of the print when there are fine details. This seems to only happen near the end of the print, not in the middle even with models with a lot of fine details (like the jungle boat).

While printing some other small detailed models I noticed extruder motor quitting mid print. The problem was a bit hard to repeat but then, during some stringing tests using the 'basic stringing test', after one successful print it started quitting exactly at the same spot, like at the same layer exactly at the same spot. 
This exact malfunctioning quite well ruled out any hw issue. Sure, if something overheats it overheats at about same rate but can't be one layer accurately repeated.

First I suspected driver overheating but took care of that with a heatsink and PLENTY of airflow. Also I suspected hw issue as usually before the failure there's a screech noise (like really bad step commands) coming from the motor, but like said, as this happens exactly at the same spot always, I'm fairly sure it's not the hw. But this would indicate it's not the S3D either as the noise indicates that there's something wrong in the command chain.

Also, I don't anymore believe it's a driver issue because if I instantly after the print fail cancel the job and try to manually extrude via octoprint UI, the extruder motor works just fine.

This is not:
- Clogging issue. The motor just stops even trying, it's fully torque off. 
- Linear advance issue as it is turned off (verified by M900 not being recognized)
- Stepper motor over heating. It's around 40C.
- Stepper driver over heating. I added a heat sink and A LOT of airflow. Haven't touched in the current adjustment so it's factory setting and as the motor isn't overheating, I doubt it's too much.

### Expected behavior

Should finish printing

### Actual behavior

extruder motor quits trying

### Steps to Reproduce

Print https://www.thingiverse.com/thing:2080224 with my gear with my settings

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.0.x

### Printer model

Tronxy x5sa Pro

### Electronics

stock v6 main board

### Add-ons

improved driver cooling

### Your Slicer

Simplify3D

### Host Software

OctoPrint

[marlin_conf.zip](https://github.com/MarlinFirmware/Marlin/files/6176255/marlin_conf.zip)
[image]
[S3D Profile Tronxy X5S - PLA - tower test failing.zip](https://github.com/MarlinFirmware/Marlin/files/6176258/S3D.Profile.Tronxy.X5S.-.PLA.-.tower.test.failing.zip)

thisiskeithb, 2021-03-20:
Does the issue occur with different slicers?

Have you tried printing directly from SD instead of over serial?

willmackey2nd, 2021-03-20:
** Fixed / removed the first paragraph from the description as it was a copy paste fault.

Just verified with this string pillar model that SD card printing failed exactly at the same spot. Have not tested other slicers (shamefully admit that I haven't never really used but S3D during my 6 printing years). I could give Cura a try but I somehow don't think it could be slicer fault, with the weird motor noise and all.

thisiskeithb, 2021-03-20:
> Just verified with this string pillar model that SD card printing failed exactly at the same spot.

Was that the exact same gcode you were printing with OctoPrint or did you re-slice it? If it’s the same, then the gcode could be could have been corrupt.

Try slicing in a different slicer and see if your printer stops in the same spot.

@rhapsodyv: Are you aware of any common issues on these Tronxy machines that could cause this?

willmackey2nd, 2021-03-20:
Yes, exactly the same file to via OctoPrint and SD card.

I installed Cura and quickly set it up closely to S3D as I could (same startup scripts, same speeds, same retraction etc.) and shot it at Marlin via OctoPrint. It succeeded.

I've heard some "rumors" about some issues with G92 E0 with Marlin. No idea if they're valid anymore and if they're only S3D related or general issue but there's an option to disable them in S3D. Didn't yet try it and will call it for the day but I'll test that tomorrow. Then I found some comments about "something something ...issues with S3D retraction and Marlin... something" but again really didn't find any details to back that statement.

[image]

willmackey2nd, 2021-03-20:
Oh, and about corrupted gcode, well could be, but I ran into this problem first with some other models too. They failed always at the very end of the model when there was some small details, so I initially thought it was extruder driver overheating or something like that, but then I started noticing they fail at exactly same position every time and then I finally found this stringing test that allowed me to make it fail fast and repeatedly, so it became obvious it couldn't be a hw issue.

rhapsodyv, 2021-03-20:
Thanks for sharing your gcode. I will do some simulations.

willmackey2nd, 2021-03-21:
Bugger. G92 E0 did not solve it. Stops at the same spot.
I took a video of it too just to show the noise of the extruder.
https://www.youtube.com/watch?v=W1gSz2q9b5E

mluckau, 2021-03-23:
I had exactly the same problem with my Tronxy X5SA-Pro. I use the PrusaSlicer as slicer. I printed exactly the same GCode again and then it worked without problems.

willmackey2nd, 2021-03-23:
This case is different as it is repeatable.

willmackey2nd, 2021-03-23:
> I had exactly the same problem with my Tronxy X5SA-Pro. I use the PrusaSlicer as slicer. I printed exactly the same GCode again and then it worked without problems.

This case is different I believe as it is repeatable.
Do you recall it making screwed up steps noise when it happened?

willmackey2nd, 2021-03-26:
@rhapsodyv any luck reproducing the issue?
I've now been able to reproduce this with various models while trying to tune the printer otherwise.
Latest was this small heat tower. I dropped the model -15mm in S3D and added some temp changes and extruder stops always at the end of layer 26 / at the beginning of layer 27. I saved the serial log of last run. Not able to spot any suspicious there either. Is there some extra debugging stuff I could turn on in Marlin to help troubleshooting this?

I've also played with different FW settings while tuning. Been chaning X/Y accelerations, switching between classic jerk and junction deviation, changed extruder max speeds / accelerations and these haven't affected this stopping behavior at all.

[serial.zip](https://github.com/MarlinFirmware/Marlin/files/6211272/serial.zip)
[Heat_tower_190-220_Bridge.zip](https://github.com/MarlinFirmware/Marlin/files/6211253/Heat_tower_190-220_Bridge.zip)
[Heat_tower_190-220_Bridge_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/6211257/Heat_tower_190-220_Bridge_gcode.zip)

willmackey2nd, 2021-03-26:
Everything I do just makes it fail more. Now the other XY motor decided to have siesta mid print. I wonder if it's just the shitty chitu board deteriorating already after a roll of filament or if this really is some FW issue.

mydewji, 2021-07-14:
I seem to be having a very similar issue. I've recompiled Marlin of the Bugfix branch and disabled both s-curve acceleration and linear advance and have the same symptoms - extruder motor just stops mid print. However interestingly if I disable retraction in Cura it works fine. (Same issue when enabling retraction in PrusaSlicer).

I have an Ender 3 Pro w/ an SKR Mini V2 board (2209 stepper drivers running Stealthchop) and a Trianglelabs DDE kit. I'm throughly stumped and this is the closest bug report I've seen similar to my issue.

willmackey2nd, 2021-07-15:
I had to switch to a duet board as I couldn't find any solution besides printing slow, which wasn't a solution for me for this printer. 

Good to know it's not a chitu board issue only. I can use that board for another slower setup.

luizmont, 2021-08-30:
> I seem to be having a very similar issue. I've recompiled Marlin of the Bugfix branch and disabled both s-curve acceleration and linear advance and have the same symptoms - extruder motor just stops [...]

Any luck solving the problem?
I have the same board and the same issue...
I just opened a new issue, it's #22676

mydewji, 2021-08-30:
Chiming in again - I finally fixed it by replacing my entire extruder stepper motor. Everything works as expected now - I'd check the wiring and try a new motor and see if it works for you.

luizmont, 2021-08-30:
Thank you for your answer mydewji.
I ordered a new stepper motor and I'll try that!

5ft24dave, 2021-09-20:
In the duplicate issue, https://github.com/MarlinFirmware/Marlin/issues/22676#issuecomment-922380049_
another user had the same issue and it was the stepper extension cable with their Micro-Swiss direct drive extruder kit.
I have the Micro-Swiss direct drive kit as well. Worked fine with the creality 4.2.7 board until I loaded Marlin 2.0.9.1 All the test gcode from the teachingtech website stops extruding after about 10-20mm. doesn't restart for the next layer etc. If I slice any STL of my own with Cura, it prints with no issues at all.  The web site says all their test gcode is sliced using simplify3D. all of the test gcode files print fine on my Anet A8, and also printed fine on the Ender 3 with the ancient stock motherboard running Marlin 1.1.9.1 I had a brand new stepper motor, so replaced the extruder motor, Adjusted the stepper VREF to 1.2V per creality's tech.  One thing I am noticing is there is no feedrate after the Extruder distance commands in the gcode. Does Marlin 2.0.x need explicit feed rates on each command instead of a global feed rate on the first one to carry it over?  Guess I answered that myself, as the next line also has a feedrate, but still doesn't extrude.
Here is a snippet:

;layer 1, Z = 0.200
T0
G92 E0.0000
G1 E-1.0 F2100 ; custom retraction
;fan1;
G1 Z0.200 F1200
G1 X67.9080 Y145.7420 F6000
G1 E0 F2100 ; custom un-retraction/prime
G92 E0.0000
G1 X67.9080 Y64.4240 E3.5052 F1800
G1 X69.4260 Y62.9060 E3.5978
G1 X156.5740 Y62.9060 E7.3542
G1 X158.0920 Y64.4240 E7.4468
G1 X158.0920 Y97.7060 E8.8814
G1 X157.6610 Y98.6950 E8.9279
G1 X124.1090 Y129.9460 E10.9044
G1 X123.1940 Y130.3060 E10.9468
G1 X122.8900 Y130.3060 E10.9599
G1 X122.8900 Y130.4930 E10.9679
G1 X122.4590 Y131.4830 E11.0144
G1 X88.0230 Y163.5570 E13.0429
G1 X84.9080 Y162.2010 E13.1894
G1 X84.9080 Y147.2600 E13.8334
G1 X69.4260 Y147.2600 E14.5007
G1 X67.9080 Y145.7420 E14.5933
G92 E0.0000
G1 E-1.0 F2100 ; custom retraction
;zhop1
G1 X68.4840 Y145.5030 F6000
G1 Z0.200 F1200
G1 E0 F2100 ; custom un-retraction/prime
G92 E0.0000
G1 X68.4840 Y64.6630 E3.4846 F1800
G1 X69.6650 Y63.4820 E3.5566
G1 X156.3350 Y63.4820 E7.2925
G1 X157.5160 Y64.6630 E7.3645
G1 X157.5160 Y97.5860 E8.7837
G1 X157.1810 Y98.3550 E8.8198
G1 X123.7960 Y129.4500 E10.7864
G1 X123.0840 Y129.7300 E10.8194
G1 X122.3140 Y129.7300 E10.8526
G1 X122.3140 Y130.3730 E10.8803
G1 X121.9790 Y131.1430 E10.9165
G1 X87.9070 Y162.8780 E12.9236
G1 X85.4840 Y161.8230 E13.0375
G1 X85.4840 Y146.6840 E13.6900
G1 X69.6650 Y146.6840 E14.3719
G1 X68.4840 Y145.5030 E14.4439
G92 E0.0000

thisiskeithb, 2023-08-07:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

willmackey2nd, 2023-08-08:
Unfortunately at least I can't test this as I've been running the printer with Duet 2 board last 2 years. Zero stopping issues during that time in case someone still thought it could be a wiring issue, but really it never really could have been if the issue was repeatable at the same exact gcode line.

closed 2023-10-19 by github-actions[bot]