MarlinFirmware/Marlin #21394: [BUG] (BLTOUCH_DELAY cause BLTouch Probing 1 more times)

NandoGommez, 2021-03-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm not sure if this is a BUG, ​​or if it really should happen. Anyway, I came here to ask in order to be helped, or maybe report a new bug and help him.

I used to compile my own marlin 2.0 for my Ender 3 printer, as I don't trust the original firmware.
I just activated the "Thermal Runway" and soon I also activated the "BLTOUCH_DELAY" and put it in 750.

Whatever settings I comment or uncomment, I test 1 by 1 to be sure.

The Bug:

Right after activating "BLTOUCH_DELAY" and set 750 (in order to improve the BLTouch recognition, to avoid failures, I'm in no hurry in probing)

Every time the printer turns on for the first time, it tests the BLTouch 2x (Self-Test), this I know, it's normal and so far so good. But when I'm going to do a homing BLTouch does a "safe-test" again.
When I give a G29, it does a "safe-test" again. 
but it stows and deploys only 1 time.

Well, I don't know if this is on purpose because I increased the delay, and there are no other problems, the leveling is perfect.
So I'm like "well, great now he does Self-Test before G28 and before G29, cool.

But was this supposed to happen?

### Bug Timeline

New bug

### Expected behavior

Just expected "normal" behavior

### Actual behavior

Do a self-test every time I give G28 or G29. If this is not a bug, well I don't mind BLTouch doing a self-test before any action, like before a G29 or G28, either way it increases the security that it doesn't there is failure

### Steps to Reproduce

1-> Active BLTOUCH_DELAY and put 750.
2-> Turn on the printer and do a self-test.
3-> But it does a self-test again every time I send G29 or G28, either via OctoPrint or via the LCD

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

Ender 3 Pro

### Electronics

Mainboard Creality 4.2.2

### Add-ons

No add-ons

### Your Slicer

Cura

### Host Software

OctoPrint

NandoGommez, 2021-03-21:
Just extra information, I disabled "USE_ZMIN_PLUG"

`//#define USE_ZMIN_PLUG`

The reason I turned it off, is that I no longer use the z-axis switch, nor is it connected to the board.
and then I activated "DETECT_BROKEN_ENDSTOP"

`#define DETECT_BROKEN_ENDSTOP`

BLTouch works normally as it is.
Just doing self-test before the actions. did not do this before enabling BLTOUCH_DELAY

NandoGommez, 2021-03-23:
does anyone have an idea?

FanDjango, 2021-03-23:
> So I'm like "well, great now he does Self-Test before G28 and before G29, cool.

> But was this supposed to happen?

No, not really. Can you produce a small video?

NandoGommez, 2021-03-23:
> > So I'm like "well, great now he does Self-Test before G28 and before G29, cool.
> 
> > But was this supposed to happen?
> 
> No, not really. Can you produce a small video?

Ah yes, here's the video.
https://www.youtube.com/watch?v=d_J3aRranEI

if I go back to the old firmware, it doesn't.

However, in any case, I already made some prints and they all came out perfect and well leveled, the leveling did not change.

NandoGommez, 2021-03-23:
> Ah yes, here's the video.
> https://www.youtube.com/watch?v=d_J3aRranEI
>
> if I go back to the old firmware, it doesn't.
>
> However, in any case, I already made some prints and they all came out perfect [...]

notice that as soon as I hit "Auto Home", he does a little self-test. and during Probing he does it again halfway.

FanDjango, 2021-03-23:
Yes, I see exactly what you describe. That's not a self test. Something is wrong. Even if it is not a catastrophe.

Now, the following would be of interest:

1. Turn on the levelling debug in the configuration.h (read the comment of that, too)
2. then dont forget to turn it on by command as described in the comment 
3. then do a G28 
4. Let's look at the output of the console, it will have debugging information for the homing

5. need to see the configuration.h and ..._adv.h

Before that, no one can help.

FanDjango, 2021-03-23:
Oh, and you are sure the behaviour goes away when you do not specify the BLTOUCH_DELAY?

NandoGommez, 2021-03-23:
> Yes, I see exactly what you describe. That's not a self test. Something is wrong. Even if it is not a catastrophe.
>
> Now, the following would be of interest:
>
> 1. Turn on the levelling debug in the [...]

Okay, I will definitely do that later as soon as I vacate.

I will upload my files "Configuration.h" and "Configuration_adv.h", Who knows, you can do an analysis and see if something is wrong, even though I just configured the basic things.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6189000/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6189001/Configuration_adv.zip)

NandoGommez, 2021-03-23:
> Oh, and you are sure the behaviour goes away when you do not specify the BLTOUCH_DELAY? Because 750 is the default anyway.

Yes, when I deactivate it, it doesn't.
Wouldn't the default be 500? I thought it was 500 so I increased it to 750 with the intention of improving the response rate.

FanDjango, 2021-03-23:
Yes, 500. Sorry

closed 2021-06-02 by github-actions[bot]