MarlinFirmware/Marlin #21395: [FR] (feature summary)

Ant05, 2021-03-21:
### Is your feature request related to a problem? Please describe.

Commonly used LCD's such as 12864 or 2004 suffers pixels damaging over time. 
In fact during printing process the display is always on, this, after some month of usage lead to damages, you can, in fact, clearly see some pixel deterioration. 
This happens for many users, and even if I changed the display several times (and the board), still, it happens.

  

### Describe the feature you want

Since this issue is probabily caused by the pixels staying on during the entire printing process we can make the LCD's lives longer if we turn them off when the user doesn't need them.
Using something like LCD.clear() all the pixels will turn off, and if the wiring makes it possible we can turn off even backlight.
So it should work something like that:
You have in the config.h file something like #define LCD_IDLE_TIMEOUT 60, this will be the time in seconds you'll wait for the LCD to perform a clear and a backlight shut down, this time is counted since the last user input (eg from the rotary encoder or buttons).
When the user comes back and presses buttons, rotary encoder or whatever, the menu will be re-printed on the screen. 

Thanks you very much for your time and your tremendous amount of great work :)

qwewer0, 2021-03-21:
https://github.com/MarlinFirmware/Marlin/issues/4728 Duplicate?

ArchTonIO, 2021-03-21:
Yes, that's the point, I'm not so interested in saving processing power but saving the LCD itself.
I read the old issue: someone said that this is not possible due to the wiring, so is not possible to turn off the backlight, but maybe is possible to perform some sort of clear() to save the pixels. am I wrong? (if this is the case tell me why I just want to know more), thanks!

thisiskeithb, 2021-03-21:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/4728

closed 2021-03-21 by thisiskeithb

mriscoc, 2021-03-26:
On LCD display there are not pixels in ON state, because the light come from the LED backlight, and the pixels are only filters. That is totally different for OLED and plasma displays, where the light comes from the pixel itself. Maybe there is some command to control the brightness of the backlight.