MarlinFirmware/Marlin #21406: [BUG] (Ramps 1.4 freeze when define HEATER_CHAMBER_PIN)

fkirita, 2021-03-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

When define HEATER_CHAMBER_PIN Octoprint connect to ramps but the board it is not responsive.

### Bug Timeline

It is a new printer.

### Expected behavior

When define HEATER_CHAMBER_PIN Octoprint connect to ramps and I want to control the chamber temperature. 

### Actual behavior

Without HEATER_CHAMBER_PIN the board works, the chamber temp it is showing in Octoprint but cannot set the chamber temperature. Just to set, not to heat the chamber (I have a passive chamber with a servo vent), the M141 response it is unknown command.
When define HEATER_CHAMBER_PIN Octoprint connect to ramps but the board it is not responsive.

### Steps to Reproduce

1. In configuration.h #define TEMP_SENSOR_CHAMBER 1.
2. In configuration_adv.h 
// Heated Chamber options
//
#if TEMP_SENSOR_CHAMBER
  #define CHAMBER_MINTEMP             5
  #define CHAMBER_MAXTEMP            60
  #define TEMP_CHAMBER_HYSTERESIS     2   // (°C) Temperature proximity considered "close enough" to the target
  #define CHAMBER_LIMIT_SWITCHING
  #define HEATER_CHAMBER_PIN          7   // Chamber heater on/off pin
  //#define HEATER_CHAMBER_INVERTING false

  //#define CHAMBER_FAN               // Enable a fan on the chamber
  #if ENABLED(CHAMBER_FAN)
    #define CHAMBER_FAN_MODE 2        // Fan control mode: 0=Static; 1=Linear increase when temp is higher than target; 2=V-shaped curve.
    #if CHAMBER_FAN_MODE == 0
      #define CHAMBER_FAN_BASE  255   // Chamber fan PWM (0-255)
    #elif CHAMBER_FAN_MODE == 1
      #define CHAMBER_FAN_BASE  128   // Base chamber fan PWM (0-255); turns on when chamber temperature is above the target
      #define CHAMBER_FAN_FACTOR 25   // PWM increase per °C above target
    #elif CHAMBER_FAN_MODE == 2
      #define CHAMBER_FAN_BASE  128   // Minimum chamber fan PWM (0-255)
      #define CHAMBER_FAN_FACTOR 25   // PWM increase per °C difference from target
    #endif
  #endif

  #define CHAMBER_VENT              // Enable a servo-controlled vent on the chamber
  #if ENABLED(CHAMBER_VENT)
    #define CHAMBER_VENT_SERVO_NR  1  // Index of the vent servo
    #define HIGH_EXCESS_HEAT_LIMIT 2  // How much above target temp to consider there is excess heat in the chamber
    #define LOW_EXCESS_HEAT_LIMIT 1
    #define MIN_COOLING_SLOPE_TIME_CHAMBER_VENT 20
    #define MIN_COOLING_SLOPE_DEG_CHAMBER_VENT 1.5
  #endif
#endif
3. in pin_ramps.h
// Temperature Sensors
//
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN                          13  // Analog Input
#endif
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN                          15  // Analog Input
#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN                        14  // Analog Input
#endif
#ifndef TEMP_CHAMBER_PIN
  #define TEMP_CHAMBER_PIN                    15  // Analog Input
#endif

### Version of Marlin Firmware

020007

### Printer model

my build

### Electronics

ramps 1.4, TMC 2208 uart, servo probe, servo vent, no lcd

### Add-ons

enclosed printer

### Host Software

OctoPrint

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6178272/config.zip)

thisiskeithb, 2021-03-21:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

fkirita, 2021-03-22:
Thank you for your response.
I have downloaded the bugfix and the issue remain.
Let me describe what I want to do: I have an enclosed printer and I want to control the chamber temperature just with the heated bed and a servo vent. 
So far the hardware it is working, the servo vent open and close the vent when I send M280 command.
I have defined the chamber in configuration.h, in configuration_adv.h and the #define TEMP_CHAMBER_PIN           15 in pins_ramps.h

With the //#define HEATER_CHAMBER_PIN I see the chamber temperature in Octoprint but when I send an M141 the terminal echo Unknown command.
If I enable #define HEATER_CHAMBER_PIN      7 (or another pin) the board freeze before checking the tmc connection.
If I set #define HEATER_CHAMBER_PIN      -1 the the board start ok but M141 Unknown command it is there.

Both in 2.0 or in bugfix.

Please help.
Thank you.

ellensp, 2021-03-22:
heated chamber needs a HEATER_CHAMBER_PIN, without defining the pin HEATER_CHAMBER is disabled. 
no number after HEATER_CHAMBER_PIN is not valid and disabled,and -1 after HEATER_CHAMBER_PIN is disabled. both result in HEATER_CHAMBER being disabled.

Analog pin 7 is digital pin 61 , ie the Y_DIR_PIN. This will conflict!

ellensp, 2021-03-22:
when HEATER_CHAMBER is enabled correctly, CHAMBER_VENT seems to cause the printer to halt

ellensp, 2021-03-22:
if you disable MOVE_SERVO(CHAMBER_VENT_SERVO_NR, 90); in Marlin/src/module/temperature.cpp it doesn't crash.  That as far as ive gotten.

fkirita, 2021-03-22:
Thank you. Please tell me how I pick digital pin 7?
Also if I want just an vented chamber, heated by bed and cooled by servo vent it is possible? Let say it is a VENTED_CHAMBER solution who is independent from HEATED_CHAMBER.

Thanks again.

fkirita, 2021-03-22:
Yes you are right.
I have selected pin 4 for heated chamber and commented use of vent and everything it is working.
If I enable servo vent the board halt.

fkirita, 2021-03-26:
Hello. Any news? Or suggestion?
Thank you.

slowbro, 2021-07-08:
I beleive this is being caused by the implementation of `safe_delay()`: it itself calls `Temperature::manage_heaters()`, which then calls `MOVE_SERVO`, which then calls `manage_heaters` ....

If I comment out the calls to `manage_heaters()`, it no longer hangs.

```diff
diff --git a/Marlin/src/core/utility.cpp b/Marlin/src/core/utility.cpp
index b810855d52..d3c01d62eb 100644
--- a/Marlin/src/core/utility.cpp
+++ b/Marlin/src/core/utility.cpp
@@ -29,10 +29,10 @@ void safe_delay(millis_t ms) {
   while (ms > 50) {
     ms -= 50;
     delay(50);
-    thermalManager.manage_heater();
+    //thermalManager.manage_heater();
   }
   delay(ms);
-  thermalManager.manage_heater(); // This keeps us safe if too many small safe_delay() calls are made
+  //thermalManager.manage_heater(); // This keeps us safe if too many small safe_delay() calls are made
 }
 
 // A delay to provide brittle hosts time to receive bytes

```

thisiskeithb, 2021-07-09:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22313

closed 2021-07-09 by thisiskeithb