MarlinFirmware/Marlin #21408: Creality 4.2.7 screen "Clicking" on Marlin 2.0.7.2 [BUG] (bug summary)

XxTWMLxX, 2021-03-21:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Downloaded Marlin 2.0.7.2 from 2.0.x repo, downloaded the matching configuration example for Ender 3 pro 4.2.7 board.
Compiled no errors and flashed. when bed is heating up the screen constantly "clicks" like the button is pressed and makes an audible click from the speaker making printer screen useless while bed is heating.

### Bug Timeline

Stock creality is Marlin 2.0.1 according to website i updated to newest 2.0.7.2 and it does it now. 

### Expected behavior

firmware to flash and bed to heat normally without effecting screen/input button

### Actual behavior

constant clicking on the "button" used to control printer and screen selecting and deselecting whatever menu entry it is on.

### Steps to Reproduce

compile 2.0.7.2 marlin with matching examples for ender 3 pro creality 4.2.7. 
flash it
heat up bed 
screen should click the currently selected option over and over again.

### Version of Marlin Firmware

2.0.7.2

### Printer model

Ender 3 pro

### Electronics

4.2.7 board

### Your Slicer

Cura

### Host Software

SD Card (headless)

thisiskeithb, 2021-03-21:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

XxTWMLxX, 2021-03-22:
Attempted to compile the bugfix 2.0.x and it compiles and gives me a .bin but when trying to flash the printer it stays blank blue screen when powered on like its trying to flash, never fully boots up with the new firmware.

XxTWMLxX, 2021-03-23:
Video of printer running bug fix 2.0.x still clicking 
https://youtu.be/tr--Dl-80pU

mcrustk2, 2021-03-28:
I have exactly the same issue with the BugFix 2.0 Branch (as of today 28 Mar 21). I just flashed from Creality stock to the Marlin 2.0 bugfix branch. I have a Ender 3 (Non Pro) and the 4.2.2 board. This pressing/clicking happens on any menu for me. Its unusable.

lilinormalgirl, 2021-03-28:
No i didnt

mcrustk2, 2021-03-28:
I just took the config files from the Hotfix branch for the Ender 3 (standard) for 4.2.2, I added several missing probe related values to and compiled against Marlin 2.0.7.2 (as 4.2.2 doesn't exist for this release) and flashed. I can report that the same clicking still exists on 2.0.7.2 (if its of note I do have a BLTouch 3.1 with the single cable connection to the 4.2.2 board and the ZStop disconnected). An observation > this only seems to happen when the printer is 'doing something', when its on but stationary this is happening less frequently (but is still happening).

mcrustk2, 2021-03-29:
Okay - I have just compiled and flashed Marlin 2.0.6.1 to my Ender3 (standard) with the Creality 4.2.2 board and I can confirm the clicking/selecting problem has gone completely and normal functionality has returned. (I used the Ender3 Pro 1.5) config files and modified the name and a few BLTouch related settings for my configuration. (the config doesn't exist currently for my setup for that version (2.0.6))

Observations:

1. This is a software issue, we need to understand what has changed between 2.0.6.1 and the current 2.0.7.2. As board standard Ender3's are being sold and now with the 4.2.2 board this problem will become more prevalent.

lilinormalgirl, 2021-03-29:
Nice Works ! Congratulations

ellensp, 2021-03-30:
There are a rash of issue emerging that are related to build platform version
ie in platformio
```
[common_stm32f1]
platform          = ststm32@~12.0
```
please try latest bugfix known to have the issue 
and update platfomio to  
```
[common_stm32f1]
platform = ststm32@~6.1

```
Does this change anything?

GhostlyCrowd, 2021-03-30:
> There are a rash of issue emerging that are related to build platform version
> ie in platformio
> 
> ```
> [common_stm32f1]
> platform          = ststm32@~12.0
> ```
> 
> please try latest bugfix known to have the issue
> and update platfomio to
> 
> ```
> [common_stm32f1]
> platform = ststm32@~6.1
> ```
> 
> Make sure this is under the [common_stm32f1] section not the [common_stm32] section
> Does this change anything?

I don't know if it's related to https://github.com/MarlinFirmware/Marlin/issues/21429  

As 2.0.7.2 didn't have the platform change in it for stm32f1.  It was still @ 6.1 Worth a shot though.

SpaceXM, 2021-04-08:
Any news about this? this bug is still there and i can confirm that this happen also on Ender 5 with silent board 4.2.7 and last bugfix relase.....any hint or solution?

thisiskeithb, 2021-04-08:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21408

samwiseg0, 2021-04-09:
> Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21408

It's a duplicate of itself? Did you mean to link to another issue?

thisiskeithb, 2021-04-09:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21204

closed 2021-04-09 by thisiskeithb

mcrustk2, 2021-04-09:
@thisiskeithb How is this a duplicate, the symptoms are very different to that of#21204?

SpaceXM, 2021-04-09:
> Duplicate of #21204

issue seems different and also Ender 5 is affected.
in my case Ender 5 start printing and after 2/3 seconds this issue happen, but printing process is still good and working as expected, you can't do nothing in meanwhile until print end or printer reboot

SpaceXM, 2021-04-09:
> > Duplicate of #21204
> 
> issue seems different and also Ender 5 is affected.
> in my case Ender 5 start printing and after 2/3 seconds this issue happen, but printing process is still good and working as expected, you can't do nothing in meanwhile until print end or printer reboot

I can confirm that the problem is related to build plate heating up i will test 2.0.6.1 as soon as i can, trying to know what change was made to put ender 4.2.7 on this bug

mcrustk2, 2021-04-10:
> I can confirm that the problem is related to build plate heating up i will test 2.0.6.1 as soon as i can, trying to know what change was made to put ender 4.2.7 on this bug

It cant be because I like the OP have this issue as soon as the machine is turned on, without even heating the bed.

mcrustk2, 2021-04-10:
> There are a rash of issue emerging that are related to build platform version
> ie in platformio
>
> ```
> [common_stm32f1]
> platform          = ststm32@~12.0
> ```
>
> please try latest bugfix known to have the [...]

Im afraid this did not fix it. I just took the lates BugFix branch and made the changes you suggested, the problem still remained. I reverted back to 2.0.6.1 and all is well..

thinkyhead, 2021-04-15:
> I reverted back to 2.0.6.1 and all is well..

If you can find the very first date that the issue appears, that should give a very strong clue to the problem. How do we quickly find that date? By repeatedly testing Marlin from the date half-way between the time the bug is absent and the time the bug is present. Those dates are now months apart. But after only 4 or 5 tests, those dates will be only days apart. GitHub helpfully provide the ability to download Marlin at any point in time from the "Commits" tab above.

SpaceXM, 2021-05-29:
no news about this? very annoying problem, I have to disconnect the display every time I print

GhostlyCrowd, 2021-05-29:
> no news about this? very annoying problem, I have to disconnect the display every time I print

I bet the recent move of some of the STMF1 to the new STM32 hal would resolve this for you.  Just have to hang inthere until a experimental environment in PIO is set up by a dev for the creality series.

andreschinreal, 2021-07-15:
I am having same issue here. It randomly clicks. Like pressing the wheel.
Ender 5 32 Bits 4.2.7 board. 
Tried:
- Marlin 2.0.x Bugfix
- Marlin 2.0.6
- Marlin 2.0.6.1
- Marlin 2.0.7
- TH3D_Unified2_CrealityV4X
- Ender-5_4.2.7_Marlin2.0.1_V1.0.1TMC2225 (original firmware from creality)

Every single one of them now present the same problem. I don't know much, but I'll make a hardware check, cleaning and reinsert everything. Couldn't be firmware.

That's very annoying. I'll try to print with the screen cable off as suggested by @SpaceXM

Nensec, 2021-08-01:
I'm having the same issue that people are describing. Once the bed is heating it starts "clicking" (sometimes even scrolling) and gets amplified in frequency when the nozzle is also heating. 

@thisiskeithb the issue you marked this is a duplicate of is not the same. Can you undo it so this issue can be looked at?

thisiskeithb, 2021-08-01:
> @thisiskeithb the issue you marked this is a duplicate of is not the same. Can you undo it so this issue can be looked at?

https://github.com/MarlinFirmware/Marlin/issues/21204 covers random screen clicks due to lack of proper grounding/EMI in the LCD cable.

We don't need multiple issues open for the same thing.