MarlinFirmware/Marlin #21409: Reality Ender 3 V4.2.2 runout filament issues

AndreaAPMS, 2021-03-21:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I connected runout filament like descriptions found online following most of videos and tutorial. I configured Marlin to trigger HIGH runout filament. System is actually not considering any changes to filament switch, filament is connected between Positive and signal.
I checke which pin is used in reality V4 config file and it seems to be pa4 that is used for other purpose I think there is issues with pin assignments. I haven't found any electrical scheme of the board, tomorrow I'l try to follow with voltmeter to check the correct pin connection.

### Expected behavior

I expect to trigger an event that is not triggered

### Version of Marlin Firmware

2.0.x bugfix

### Printer model

reality ender 3 v4.2.2

### Add-ons

RUNOUIT filament switch

ellensp, 2021-03-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-03-22 by ellensp

AndreaAPMS, 2021-03-22:
ok sorry for the mistake

ellensp, 2021-03-22:
```
//
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN                    PA4   // "Pulled-high"
#endif
```

This pin is pulled high. when pin is disconected it is high and when connected it to vcc it will be high. You need to make it so you filament runout sensor connects gnd and signal.