MarlinFirmware/Marlin #21412: [BUG] TMC2209 QUICK_HOME does not respect SENSORLESS_BACKOFF_MM

georgekucher, 2021-03-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

X and Y axis does not do SENSORLESS_BACKOFF_MM before diagonal move (QUICK_HOME).
In case X or Y axis is in its min position, noisy bump happens.
Moreover, it causes the other axis (which is still moving) to do a backoff and then continue homing.
Looks like this initial diagonal move is not a HOMING itself. Actual homing starts after this diagonal move.

### Expected behavior

SENSORLESS_BACKOFF should be triggered BEFORE diagonal move as well as before actual homing

### Actual behavior

SENSORLESS_BACKOFF is NOT triggered BEFORE diagonal move

### Steps to Reproduce

Do Auto Home

### Version of Marlin Firmware

2.0.7.2, bugfix-2.0.x

### Printer model

Ender 3

### Electronics

BTT_SKR_E3_DIP_v1.1

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6180393/Config.zip)

Portzal, 2021-03-24:
I have no solution, but I can confirm that I have also noticed this behavior. For example if Y homes first it is held  on the end stop until X homes resulting in a poor homing experience.

georgekucher, 2021-03-24:
@Portzal 
Try uncomment #define HOMING_BACKOFF_POST_MM { 2, 2, 2 }
This should help.
My issue is a bit different.

Portzal, 2021-03-26:
That setting was already enabled and was ignored until X homed. I tried many different settings without success. It was simpler to just not use quick home.

georgekucher, 2021-05-26:
Is this gonna be fixed someday?

ellensp, 2021-05-26:
give this a try, untested other than compiling. 
```DIFF
diff --git a/Marlin/src/gcode/calibrate/G28.cpp b/Marlin/src/gcode/calibrate/G28.cpp
index a71f541593..ee7c8ec5c2 100644
--- a/Marlin/src/gcode/calibrate/G28.cpp
+++ b/Marlin/src/gcode/calibrate/G28.cpp
@@ -94,6 +94,17 @@
             || tmc_enable_stallguard(stepperY2)
           #endif
       };
+
+      //
+      // Quick back away to prevent an early X/Y sensorless trigger
+      //
+      #if DISABLED(DELTA) && defined(SENSORLESS_BACKOFF_MM)
+        const xy_float_t backoff = SENSORLESS_BACKOFF_MM;
+        const float x_backoff_length = TERN0(X_SENSORLESS, -ABS(backoff[X_AXIS]) * x_axis_home_dir);
+        const float y_backoff_length = TERN0(Y_SENSORLESS, -ABS(backoff[Y_AXIS]) * Y_HOME_DIR);
+        if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPAIR("Quick X Sensorless backoff: ", x_backoff_length, "mm. Quick Y Sensorless backoff: ", y_backoff_length, "mm");
+        do_blocking_move_to_xy(x_backoff_length, y_backoff_length, fr_mm_s);
+      #endif
     #endif
 
     do_blocking_move_to_xy(1.5 * mlx * x_axis_home_dir, 1.5 * mly * Y_HOME_DIR, fr_mm_s);```

closed 2021-08-06 by github-actions[bot]