MarlinFirmware/Marlin #21419: [BUG] (Mesh leveling ender 4.2.2 not working)

eep8freak, 2021-03-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I use manual mesh leveling because my bltouch hasn't arrived yet. I can set the mesh as planned and also see the Matrix, but the actual compensation is not working. I using the ender 4.2.2 Mainboard.
I added the M420 S1 command after G28 in Cura, also tried G29 instead and a combination of both, not working.
When I use M111 S32, bed leveling ist on.

### Expected behavior

Automatic height compensation

### Actual behavior

No compensation

### Steps to Reproduce

Enable manual bed leveling
Set the correct heights
Start the print

### Version of Marlin Firmware

Bugfix 2.0.X

### Printer model

Creality Ender 3

### Electronics

Mainboard 4.2.2

### Your Slicer

Cura

### Host Software

SD Card (headless)

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/6185948/Config.zip)

ellensp, 2021-03-23:
when you turn the machine on and send a G29 S0 does it show a valid mesh?

thisiskeithb, 2021-03-23:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

ellensp, 2021-03-23:
Just tried this on the sim. 
PROBE_MANUALLY and MESH_BED_LEVELING enabled
bed tilt -5 on left +5 on right.

manual mesh set with
G29 S3 I0 J0 Z-5
G29 S3 I0 J1 Z-5
G29 S3 I0 J2 Z-5
G29 S3 I1 J0 Z0
G29 S3 I1 J1 Z0
G29 S3 I1 J2 Z0
G29 S3 I2 J0 Z5
G29 S3 I2 J1 Z5
G29 S3 I2 J2 Z5

mesh saved M500
M420 S1 sent.
load in test gcode and print.

Watch raw Z kinematics move z up and down as expected to compensate. 100% working in sim.

eep8freak, 2021-03-23:
I tried the process again. I increased the absolute values a bit and saw the z-axis moving. I tuned the grid after M502. I think I have to increase the probing points due to my uneven bed.
Thanks for your help. Not a software bug, just a user error🙈

closed 2021-03-23 by eep8freak