MarlinFirmware/Marlin #21423: [BUG] CUSTOM_MENU_MAIN for LVGL UI

Malderin, 2021-03-23:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Board: MKS robin nano v.2
Last marlin 2.x bugfix

After activating CUSTOM_MENU_MAIN in configuration_adv in the "More" menu, no commands appear except for G-Code

### Version of Marlin Firmware

Bugfix 2.x

### Electronics

Mks Robin nano v2

Yetreyert, 2021-05-04:
happens to me too, were you able to solve?

Malderin, 2021-05-04:
> happens to me too, were you able to solve?

No

Valnar8, 2021-06-19:
Still no solution?  
I found out that there is a menu_main.cpp that contains seemingly crucial stuff for those menus but everything there is under a #if HAS_LCD_MENU. And since there is no such thing as HAS_LCD_MENU in either of the configs whatsoever, everything is grey.

javicarrera, 2021-07-30:
I think I have found the reason for this problem.
With this commit #18177 definitions for custom user menus were changed in Marlin.

#define CUSTOM_USER_MENUS 
#define USER_DESC_1 ""
#define USER_GCODE_1 ""
was changed by:  
#define CUSTOM_MENU_MAIN
#define MAIN_MENU_ITEM_1_DESC ""
#define MAIN_MENU_ITEM_1_GCODE ""

It is missing however changing those in conditionals_adv.h (CUSTOM_MENU_MAIN was changed, the other 2 not):
#if BOTH(HAS_TFT_LVGL_UI, CUSTOM_MENU_MAIN)
  #define _HAS_1(N) (defined(USER_DESC_##N) && defined(USER_GCODE_##N))

Please if someone can change this line of code and I think the "more" menu will work again with LVGL UI.

thinkyhead, 2021-07-31:
That is definitely the thing! Also, Marlin now provides two different menus, but only MarlinUI cares about both of them. The rest use the MAIN one, which is fine. Maybe in the future other screens will also handle `CUSTOM_MENU_CONFIG`.

closed 2021-07-31 by thinkyhead

Valnar8, 2021-07-31:
> It is missing however changing those in conditionals_adv.h (`CUSTOM_MENU_MAIN` was changed, the other 2 not):
> 
> ```c++
> #if BOTH(HAS_TFT_LVGL_UI, CUSTOM_MENU_MAIN)
>   #define _HAS_1(N) (defined(USER_DESC_##N) && defined(USER_GCODE_##N))
> ```
So what do I need to change it to?

thisiskeithb, 2021-07-31:
> So what do I need to change it to?

You can apply the changes from https://github.com/MarlinFirmware/Marlin/commit/35b0083dfee0f5508d1ecfd4756ebcac8892b067 directly, or download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) which includes the patch.

commit 35b0083d (2021-07-31): 🐛 Fix custom menus on TFT LVGL

Fixes #21423. Regression from #18177.
commit 98119166 (2021-08-09): 🐛 Fix custom menus on TFT LVGL

Fixes #21423. Regression from #18177.