MarlinFirmware/Marlin #21424: [FR] Use compressed font (format 1) for u8glib

ramiropolla, 2021-03-23:
### Describe the feature you want

u8glib supports a compressed font format (format 1) since I don't know when.

It could be used to generate the fonts in fontdata_ISO10646_1.h, which saves 312 bytes for the non-extended font and 600 bytes for the extended font.

Unfortunately I don't know how to check with the preprocessor if the u8glib version being used supports this feature or not, so I don't know if we can unconditionally change it.

This is the patch that I used for my Anet V1.0 and all seems to be working well font-wise.

```diff
diff --git a/buildroot/share/fonts/genallfont.sh b/buildroot/share/fonts/genallfont.sh
index 66f8e2c84f..f3d3b7be5f 100755
--- a/buildroot/share/fonts/genallfont.sh
+++ b/buildroot/share/fonts/genallfont.sh
@@ -93,8 +93,8 @@ done
 if [ 1 = 1 ]; then
   DN_WORK=`mktemp -d`
   cd ${DN_WORK}
-  ${EXEC_BDF2U8G} -b 1 -e 127 ${FN_FONT} ISO10646_1_5x7 tmp1.h >/dev/null
-  ${EXEC_BDF2U8G} -b 1 -e 255 ${FN_FONT} ISO10646_1_5x7 tmp2.h >/dev/null
+  ${EXEC_BDF2U8G} -b 1 -f 1 -e 127 ${FN_FONT} ISO10646_1_5x7 tmp1.h >/dev/null
+  ${EXEC_BDF2U8G} -b 1 -f 1 -e 255 ${FN_FONT} ISO10646_1_5x7 tmp2.h >/dev/null
   TMP1=$(cat tmp1.h)
   TMP2=$(cat tmp2.h)
   cd - >/dev/null
```

phizev, 2021-03-26:
I'm not sure if this helps clarify anything at all, but the source for bdf2u8g mentions [format 1 all the way back in 2011](https://github.com/olikraus/u8glib/commit/02b7fff93daf9c4de1439fa4f91c8f350c750862#diff-49933e6480bcc88c74b271392c92930b9a5cef733d49baffee6813ac3a35a23bR103).

Judging by the code in the [U8glib_Arduino library](https://github.com/olikraus/U8glib_Arduino/blob/master/src/clib/u8g_font.c#L350), I think this change looks promising with little chance of a downside.

ellensp, 2021-03-26:
This need tested across all platforms. Marlin uses a highly optimised version of this library https://github.com/MarlinFirmware/U8glib-HAL not the original code.

phizev, 2021-03-28:
Reading through [U8glib-HAL](https://github.com/MarlinFirmware/U8glib-HAL), it shares all of the same code for getting the fonts, that appears to be hardware independant. The HAL changes look limited to the hardware related code, though this is my first rodeo in looking at Marlin.

@ellensp I completely agree with you, though I have no access to anything other than RAMPS.

ramiropolla, 2021-03-29:
Would it be possible to integrate U8glib-HAL into Marlin (merging the repositories) so that we are always certain to have a version that supports format 1? There is also room for more optimisations in U8glib-HAL (I managed to save a few hundred bytes).

phizev, 2021-03-29:
@ramiropolla Format 1 was added to U8glib before 2011, as this is when it was moved to github and it is already present in the repo at that point. U8glib-HAL was started in 2017/2018, I think. In fact, I have yet to find a version of U8glib which lacks format 1 support. I don't think that the issue is whether format 1 is supported, or not, rather that it needs testing. (I stand to be corrected though.)